bluesky/ophyd #47: DOC: Propose to use google style doc-strings for ophyd

stuwilkins, 2015-01-13:
See [Google style docs](http://sphinxcontrib-napoleon.readthedocs.org/en/latest/example_google.html)

tacaswell, 2015-01-13:
Why not numpydoc?

stuwilkins, 2015-01-13:
\<personal_opinion>
I think that the google style is more compact and better read from `ipython` using the `?` magic and just more pleasing on the eye. I don't like the underlining in the numpy style.
\</personal_opinion>

However it is really @dchabot and @klauer's call as they are the architects, and we shouldn't probably [bikeshed](http://bikeshed.com) this too much. I do however think that user_api type functions need more than the usual terse string to be useful and these will *face* the general user.

klauer, 2015-01-15:
Personally, I prefer the look of google docstrings. I made a quick (and *far* from perfect) auto-conversion script for both google/numpy styles so we can all decide:

[google style branch](https://github.com/NSLS-II/ophyd/tree/docstrings)
[numpy style branch](https://github.com/NSLS-II/ophyd/tree/docstrings_numpy)

tacaswell, 2015-01-15:
The argument in favor of numpydoc is that it is the nominal standard of the scientific stack (numpy, scipy, most of the scikits-*, matplotlib (well, in the middle of a very slow conversion), pandas, ...) and is what we are using for the everything in Nikea.  There is value in reducing the cognitive load of switching between things like this.

I have been staring at numpydoc strings for several years now and I find the google style to be really cramped.  It just looks like a wall of text that I have to read, where as I am used to the white space patterns, and can jump around and answer things like 'what is the expected type of the third argument?' very quickly (find the third ':' down in the un-indented lines in the Parameters section).  

I suspect that the features of numpydoc that I find most appealing are the exact same features that make it seem visually less pleasing :smiling_imp:

dchabot, 2015-01-16:
> On Jan 15, 2015, at 12:02 PM, Thomas A Caswell <notifications@github.com> wrote:
> 
> The argument in favor of numpydoc is that it is the nominal standard of the scientific stack (numpy, scipy, most of the scikits-*, matplotlib (well, in the middle of a very slow conversion), pandas, ...) and is what we are using for the everything in Nikea. There is value in reducing the cognitive load of switching between things like this.

To assume a technical argument here, the numpydoc Sphinx extension is not available in the Debian Stable repos, thus complicating the packaging requirements.

> I suspect that the features of numpydoc that I find most appealing are the exact same features that make it seem visually less pleasing

I suspect you may be right ;-)

tacaswell, 2015-01-17:
You can install numpydoc through either pip or conda.

stuwilkins, 2015-01-25:
Hi all,

Ok after reviewing the conversation, I agree with @tacaswell on the arguments about numpy style docs. 

So.... As CTO I am deciding that ALL NSLS-II doc-strings will follow the Numpy style. Please follow this from now on across both the analysis and acquisition stacks. 

Having consistency across the data collection and analysis I believe is also important and maintaining a standard is also very important. 

Please make sure that from now on all docstrings follow the numpy style [Numpy Docs](https://github.com/numpy/numpy/blob/master/doc/HOWTO_DOCUMENT.rst.txt)

Installation of either [numpydoc](https://pypi.python.org/pypi/numpydoc) or [napoleon](https://pypi.python.org/pypi/sphinxcontrib-napoleon) is easy with conda or pip, and I don't see this a problem.

And please, no painting the bike shed ....

closed 2015-01-25 by stuwilkins