bluesky/ophyd #49: ENH: Triggering of detector classes.

stuwilkins, 2015-01-13:
Hi, @dchabot, @klauer,

I have been looking at 2 issues. 1 Time-stamping of data and secondly triggering detectors. Currently in the run_engine it is only possible to trigger a single `EpicsSignal` (it uses the put method). This doesn't allow the nice new detector classes to be used such as `EpicsScaler` or the area detector zoo. 

I wonder if it is time to start thinking about adding a `trigger()` method to these type of detectors which could be called by the run_engine. I would also suggest at this time having a `trigger_timestamp` property which is the timestamp of the trigger. Even `EpicsSignals` could have this which would just do a `self.put(1)`. Thoughts?

I am also starting to think about timestamps. It appears that we have 2 different type of detectors. Those that require triggering and have been done so in a step scan. We also have things that get added to the run which are asynchronous. It would appear that we need to distinguish between these two. I think that an `event` should always have a timestamp and contain only data which was at that timestamp, with the caveat that software triggers still group them together in one event descriptor. 

I would propose the following:

We have 3 types of *event*:

1) In a step scan there is always a trigger event, taken from the system clock which is executed just before **ophyd** starts triggering and at the end of reading detectors. This event would look like:
```
trigger {
    start : (float of start from time.time())
    stop : (float of stop)
}
```

This is useful in data collection because it is the execution window of the step scan. 

For detectors that are triggered, they are grouped into a single *event descriptor* which would look like:
```
detector_group_1 {
    pos1 : (value of motor)
    sclr_1 : (scalar channel 1)
    ...
    timestamp : (earliest trigger value)
}
```

The timestamp would be taken from the minimum value of the trigger timestamps (there is no good way to do this.

Lastly, `EpicsSignals` which are not triggered are recorded as separate events with there own timestamp:
```
temp_a {
    temp_a : (value)
    timestamp : (earliest trigger value)
}
```

I think this could be achieved with a little config. This pushes some work onto the data side but provides the truest representation of what actually happened without having each detector have a single event. I don't know if @tacaswell, @ericdill or @danielballan think this is what we discussed *at length* yesterday. 

Also, this could be configurable, with the option to just record each detector as its own event as an advanced option.

While this may not be the long term goal of the run_engine. I wonder if this could be tested (I would be happy to implement a test branch) so that we can see how this runs and what the pitfalls on the data collection side are. Thoughts?

dchabot, 2015-01-13:
> On Jan 13, 2015, at 1:14 PM, Stuart Wilkins <notifications@github.com> wrote:
> 
> Hi, @dchabot, @klauer,
> 
> I have been looking at 2 issues. 1 Time-stamping of data and secondly triggering detectors. Currently in the run_engine it is only possible to trigger a single EpicsSignal (it uses the put method). This doesn't allow the nice new detector classes to be used such as EpicsScaler or the area detector zoo.
> 
> I wonder if it is time to start thinking about adding a trigger() method to these type of detectors which could be called by the run_engine. I would also suggest at this time having a trigger_timestamp property which is the timestamp of the trigger. Even EpicsSignals could have this which would just do a self.put(1). Thoughts?

Ken and I had discussed the trigger problem some weeks ago.
At the time, I suggested an adapter class for ‘detector-like’ objects:

--------------------------------------------------
class Detector(object):

    def __init__(self, detector_thing):
        self._detector = detector_thing

    def acquire(self, **kwargs):
        '''Configure and/or actuate the detector to affect acquisition.
           May be regarded as a "trigger"

           Called by the RunEngine.
           Returns None.
        '''
        pass

    def read(self, **kwargs):
        '''Retrieve data from instrumentation, format it, and return it.

           Returns dict: {data_key: value} or {data_key: [val0, val1,...]}
        '''
        pass
-------------------------------------------------

One of the ideas being that the life-cycle of these ‘detector_things’ need only be coincident with the duration of a scan (i.e. they need only exist transiently, as I’m not sure where else these capabilities are useful).
The other idea being customization of “read( )” return values...

> It appears that we have 2 different type of detectors. Those that require triggering and have been done so in a step scan. We also have things that get added to the run which are asynchronous. It would appear that we need to distinguish between these two.

By asynchronous(ly) recorded data, are you talking about CA Archiver data that may be referenced after the fact?
If so, then as long the “two detector types” share a common time-source (in our case, they do!), I don’t see a need for distinction.

If not, then I think the distinction is artificial (arbitrary?).

> I think that an event should always have a timestamp and contain only data which was at that timestamp, with the caveat that software triggers still group them together in one event descriptor.
>
> I [...]

Do you mean “useful in data analysis”?
Other than maintaining statistical info for profiling purposes, why would this be useful to data collection?

> For detectors that are triggered, they are grouped into a single event descriptor which would look like:
>
> detector_group_1 {
>     pos1 : (value of motor)
>     sclr_1 : (scalar channel 1)
>     ... [...]

Some of these ideas go back to the earliest representation of the RunHeader contents - all recorded items had a timestamp, value, and (maybe) a PV name:

data : {
		var_x : { pv: “some:pv:name.VAL”, timestamp: 123456.789, value: 987.654 }
}

I think I would like to discuss these ideas in person when I get back to BNL on Thursday afternoon.
I am currently re-working the RunEngine internals, so the proposed changes are conflict-bound. Please hold off.

— dc=

stuwilkins, 2015-01-13:
> Hi, @dchabot, @klauer,
>
> I have been looking at 2 issues. 1 Time-stamping of data and secondly triggering detectors. Currently in the run_engine it is only possible to trigger a single EpicsSignal [...]

Ken and I had discussed the trigger problem some weeks ago.
At the time, I suggested an adapter class for ‘detector-like’ objects:

--------------------------------------------------
class Detector(object):

def __init__(self, detector_thing):
self._detector = detector_thing

def acquire(self, **kwargs):
'''Configure and/or actuate the detector to affect acquisition.
May be regarded as a "trigger"

Called by the RunEngine.
Returns None.
'''
pass

def read(self, **kwargs):
'''Retrieve data from instrumentation, format it, and return it.

Returns dict: {data_key: value} or {data_key: [val0, val1,...]}
'''
pass
-------------------------------------------------

One of the ideas being that the life-cycle of these ‘detector_things’ need only be coincident with the duration of a scan (i.e. they need only exist transiently, as I’m not sure where else these capabilities are useful).
The other idea being customization of “read( )” return values...

> It appears that we have 2 different type of detectors. Those that require triggering and have been done so in a step scan. We also have things that get added to the run which are asynchronous. It would appear that we need to distinguish between these two.

By asynchronous(ly) recorded data, are you talking about CA Archiver data that may be referenced after the fact?
If so, then as long the “two detector types” share a common time-source (in our case, they do!), I don’t see a need for distinction.

The example I think of is adding an EpicsSignal as a detector which is not “triggered” This will have a value and timestamp, but it won’t be triggered each datapoint even if it is queried at each datapoint. It is then important to have the timestamp available so that during data analysis this can be correctly dealt with.

While at a base implementation the distinction is kinda arbitrary, there is a difference between detectors that we force trigger (such as scalers and areDetectors) and perhaps signals which are on a scan loop in an IOC. How we treat them during recording is important.

If not, then I think the distinction is artificial (arbitrary?).

> I think that an event should always have a timestamp and contain only data which was at that timestamp, with the caveat that software triggers still group them together in one event descriptor.
>
> I [...]

Do you mean “useful in data analysis”?
Other than maintaining statistical info for profiling purposes, why would this be useful to data collection?

This is useful because in a step scan it gives data analysis the window over which all detectors were measured. This is really useful after the fact if you try to correlate with other things. We should discuss this in person when you are back.

> For detectors that are triggered, they are grouped into a single event descriptor which would look like:
>
> detector_group_1 {
> pos1 : (value of motor)
> sclr_1 : (scalar channel 1)
> ...
> timestamp : [...]

Some of these ideas go back to the earliest representation of the RunHeader contents - all recorded items had a timestamp, value, and (maybe) a PV name:

data : {
var_x : { pv: “some:pv:name.VAL”, timestamp: 123456.789, value: 987.654 }
}

I always really liked this… that each value gets its timestamp which makes the distinction go away. I think this is looking at two sides of the same thing. lets discuss in person Thursday.

I think I would like to discuss these ideas in person when I get back to BNL on Thursday afternoon.
I am currently re-working the RunEngine internals, so the proposed changes are conflict-bound. Please hold off.

Holding off………...

— dc=

klauer, 2016-03-03:
Closing as this is no longer relevant with the new `Device` hierarchy. Custom trigger functions can be implemented, multiple timestamps can now be recorded, etc.

closed 2016-03-03 by klauer