bluesky/ophyd #68: Sort out scan_ids

ericdill, 2015-02-17:
The mechanism for producing scan_ids needs to be investigated.  As of now, the scan_id seems to reset to an old number on CSX.  The behavior should be that the scan_id defaults to something like 

```
   session_manager.scan_id = int(DataBroker[-1].scan_id) + 1
```

On ophyd startup go grab the last scan_id from `DataBroker`.   This scan_id can be overridden by the user at run time.

Reported by Claudio at CSX

arkilic, 2015-02-17:
I think we could do this within metadatastore via mongo daemon. Mongo provides bunch of aggregation methods that would do this much faster than us. We could use $max. This would return the maximum scan_id value given an owner or w/e query we are interested in. Much simpler than getting bunch of headers and unpacking to get to desired info.

dchabot, 2015-02-17:
> On Feb 17, 2015, at 4:45 PM, Eric Dill <notifications@github.com> wrote:
> 
> The mechanism for producing scan_ids needs to be investigated. As of now, the scan_id seems to reset to an old number on CSX. The behavior should be that the scan_id defaults to something like
> 
>    session_manager.scan_id = int(DataBroker[-1].scan_id) + 1
> 
> On ophyd startup go grab the last scan_id from DataBroker. This scan_id can be overridden by the user at run time.
> 
> Reported by Claudio at CSX

I don’t want that degree of coupling between ophyd and DataBroker.
Let’s investigate why ophyd is producing incorrect numbers and simply fix that…

People can and will change the scan_id as they see fit, regardless of where it ends up downstream…

stuwilkins, 2015-02-18:
+1 Daron. 

Just to be clear again: 

1) `scan_id` are not unique and there is no guarantee that they are. They are a convenience method to assign a number to scans and it is a *convinience* that they are incremented by `ophyd`. They are just a human readable concept. 
2) The use should be able to define the behavior, or even reset or advance the number depending on their needs.
3) I 100% Agree with Daron that we don't want this polluting between MDS and `ophyd`
4) We should fix the problem at source.
5) To my knowledge this has happened once, lets focus on things that really need fixing. 
6) This may be a non-issue, at one point we had the test stand putting log entries into the CSX logbook. That caused some confusion as the number is therefore out of sequence. 

This does have some implications for the new broker interface, there should probably be some way of defining a default search terms before using the indexing notation such as (in pseudo code):
```
db = DataBroker()
db.set_default_search(begin_time=xxx, end_time=xxx, experiment=xxx)
events = db[10]
```
This way you can always fence the search so that the indexing works as you intended. I have even been thinking that we may reset scan_id daily during commissioning, or weekly (every tuesday)

ericdill, 2015-02-18:
I agree with @stuwilkins and @dchabot .

To clarify the original issue (Re #5-6), Scrolling back through the log book at CSX, there are a number of places where the scan id seems to advance for ~50 scans, reset to an earlier number, advance by +1 for another ~50 scans, jump to 10,000, reset back to 6500, jump to 6800, etc...

stuwilkins, 2015-02-24:
One suggestion I have, is that we could hook the storing of these scan_id from python to the actual python prompt. The current problem is that the store is only written on exit so if python crashes or is nuked by a kill -9 you loose all the counting you did. I think this would be very easy and a more robust way to do this.

dchabot, 2015-03-21:
Still seeing scan_id's repeating (at 28id).

This seems strongly correlated with folks opening multiple ipython (ophyd) sessions - one to perform scans and another to examine data from previous/current scan. This is a perfectly reasonable thing to do...

Ex: Pavol was running scan_id==134, opened another ipython terminal and was told the current scan_id was 131. It's easy to see how this gets out of sync: simply start a scan from a terminal *different* from the one that was previously used. *Why* this gets out of sync is harder to explain. Perhaps there is zero locking on `%store`'d variables?

Options:
==============
1) strongly couple ophyd to MDS for storage/retrieval of scan_id. Boo....
2) take care of locking/unlocking access to scan_id in ophyd.
3) others??

stuwilkins, 2015-03-22:
I wonder if the solution could be to not have multiple sessions. While starting a second ipython session is very desirable, there is no need to have the controls etc. tied to this (it doesn't need to be an ophyd session). 

It also seems to me like the scan_id is also coupled to a run_engine. A run_engine is a singleton and therefore scan_id should increment for each run_engine. Having multiple run_engines is OK with the scary point that the hardware better not be the same or there is a fight for who has the steering wheel. Having a mechanism for many scan engines per beamline may be desirable but perhaps this should be denied. This could be done by a locking mechanism? 

Perhaps one way round this is to have specific ophyd sessions which don't use the ipython standard config (for example by typing `ophyd` at the prompt) which means that only that process has the scan_id and other saved state. We then setup normal ipython for analysis which just has DataBroker^H^H^H^H^H^H Portal and the other analysis side routines as standard. 

Thoughts? I vote for decoupling ophyd from the ipython standard config.

gjwillms, 2015-03-23:
I vote for decoupling, but not enforcing unique ophyd sessions.

As an operational issue, I tend to think that it is very important to maintain the ability to have multiple ophyd sessions.  If I log in from off-site to fix a problem, I should be able to do so without interfering with what the users are doing locally.  SPEC does have a feature that allows multiple users to access a single control session, which is very useful in debugging problems remotely, but I doubt that we'll reimplement that in the near future.

Having said that, SRX already has an 'ophyd' shell script that wraps the "master" scanning session profile and gives it a distinctive prompt label, 'Ophyd.'  I've noticed a tendency to always just start this session--probably because of the slightly arcane command line invocation--so SRX will have the equivalent of a data analysis/goofing around profile very soon and I vote to enforce the proper use sociologically.

pavoljuhas, 2015-03-26:
As a side note, even if scan_id-s behaved as intended, IPython `%store` magic is not a good place for inter-session persistence of ophyd variables.  To my understanding, the `%store` is intended as a user scratchpad, so the users should be able to wipe it with `%store -z` without any unexpected side effects.

I am not sure why is it so inadvisable to store the scan_id-s in DataBroker.  The ophyd has to have some write access to the broker at the first place to produce records with confusing repetitive scan_id-s.  Perhaps a way out is to use some light-weight coupling.  If the broker is installed and alive, ophyd should fetch the scan_id from the broker.  Otherwise it can just reset the scan_id to 0 for the dead broker does not write.

tacaswell, 2015-03-26:
But `scan_id` _is_ a user-specific number.

Maybe the best route here is to give up on inter-session tracking of this and have `scan_id` reset to 0 when ophyd is imported.  If the user wants to continue with another session where they left off it is their job to type in the number they want to start with.

In production people are probably not going to be opening and closing the ipython session very often.

pavoljuhas, 2015-03-26:
Users already have a plenty of things to track changing their samples, planning out temperature scans, analyzing intermediate results.  It is unhelpful to use their bandwidth on a consistent increase of integer index, possibly at 5AM; that should be a job of the experiment control software.

Last experiment we were restarting the ipython-ophyd session quite often and also running several sessions at the same time to check the previous results.  If a restart were very rare, it would only make it more likely the users would forget about setting the scan_id.

dchabot, 2015-03-26:
> On Mar 26, 2015, at 2:50 PM, Pavol Juhas <notifications@github.com> wrote:
> 
> Users already have a plenty of things to track changing their samples, planning out temperature scans, analyzing intermediate results. It is unhelpful to use their bandwidth on a consistent increase of integer index, possibly at 5AM; that should be a job of the experiment control software.
> 
> Last experiment we were restarting the ipython-ophyd session quite often and also running several sessions at the same time to check the previous results. If a restart were very rare, it would only make it more likely the users would forget about setting the scan_id.

There is some fundamental misunderstanding(s) here.

If you want to run scans or interact with hardware, use “ipython —profile=xf<cell>_ophyd”, or however you’ve aliased it locally.
Please open only one of these.

If you want to look at data, analyze, etc, use “ipython”. Open as many of these as you like.

Make 2 aliases (assumes you’re in the conda ‘ophyd’ env):
1) ophyd —> “ipython —profile=xf<cell>id_profile —ipython-dir=/home/xf<cell>id/ipython_ophyd”
2) broker —> ipython -c “from dataportal.broker import DataBroker as db” -i

In the short term, that is the best I can offer.

— dc

gjwillms, 2015-03-27:
As an update, SRX has two standard ways of invoking ipython:  'ophyd' and 'srxpython'.  The former is a full, standard ophyd session.  The latter is intended for non-destructive querying and data investigation.  At the moment, I allow for motion control, but not detector readback in this session (since Count is a single point Scan).  This is tentative, but there are good reasons for allowing inspection of the instrument while a scan is on-going.

pavoljuhas, 2015-03-27:
@dchabot - I agree it might be necessary to avoid multiple ophyd sessions.  Nevertheless, the scan_id persistence does not work well, even if we had infallible users who'd never open a second ophyd.  Users may kill a frozen session, close the terminal window or just run `%store -z` and their scan_id would go haywire.

If it is so out of question to talk to the broker, please, use a pickle or text file for storing the scan_id-s instead of IPython store.

commit ebf27d86 (2015-03-10): Merge branch 'persist_scan_id'

closes #68
closed 2015-09-29 by danielballan