bluesky/ophyd #70: Bug in ophyd mesh scan

ericdill, 2015-02-18:
Command + stack trace:
```python
In [17]: dscan([[theta], [delta]], [[-.1], [-.1]], [[.1], [.1]], [2, 4])
  File "build/bdist.linux-x86_64/egg/ophyd/userapi/scan_api.py", line 285, in run
    self.check_paths()
  File "build/bdist.linux-x86_64/egg/ophyd/userapi/scan_api.py", line 196, in check_paths
    pos.name, p.low_limit, p.high_limit))

Moving positioners back to start positions....... Done.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-17-8c61004e6277> in <module>()
----> 1 dscan([[theta], [delta]], [[-.1], [-.1]], [[.1], [.1]], [2, 4])

/home/xf23id1/anaconda/envs/ophyd/lib/python2.7/site-packages/ophyd-0.0.2-py2.7.egg/ophyd/userapi/scan_api.pyc in __call__(self, positioners, start, stop, npts, **kwargs)
    535         """
    536         self.setup_scan(positioners, start, stop, npts, **kwargs)
--> 537         self.run(**kwargs)
    538 
    539     def setup_scan(self, positioners, start, stop, npts, **kwargs):

/home/xf23id1/anaconda/envs/ophyd/lib/python2.7/site-packages/ophyd-0.0.2-py2.7.egg/ophyd/userapi/scan_api.pyc in run(self, **kwargs)
    283 
    284         with self:
--> 285             self.check_paths()
    286             self.setup_detectors(self.detectors)
    287             self.setup_triggers(self.triggers)

/home/xf23id1/anaconda/envs/ophyd/lib/python2.7/site-packages/ophyd-0.0.2-py2.7.egg/ophyd/userapi/scan_api.pyc in check_paths(self)
    194                     raise ValueError('Scan moves positioner {} \
    195                                      out of limits {},{}'.format(
--> 196                                      pos.name, p.low_limit, p.high_limit))
    197 
    198     def __enter__(self):

AttributeError: 'numpy.float64' object has no attribute 'low_limit'
```

dchabot, 2015-02-18:
Thanks.
Line 196 from the trace should be:
pos.name, pos.low_limit, pos.high_limit

Patch is on the way...

dchabot, 2015-02-18:
Please test this again, and close issue if all is well.

ericdill, 2015-02-18:
same problem...

```python
In [2]: dscan([[theta], [delta]], [[-.1], [-.1]], [[.1], [.1]], [2, 4])
kwargs in the run function: {}
  File "ophyd/userapi/scan_api.py", line 286, in run
    self.setup_detectors(self.detectors)
  File "ophyd/userapi/scan_api.py", line 196, in check_paths
    pos.name, pos.low_limit, pos.high_limit))

Moving positioners back to start positions....... Done.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-8c61004e6277> in <module>()
----> 1 dscan([[theta], [delta]], [[-.1], [-.1]], [[.1], [.1]], [2, 4])

/home/xf23id1/Repos/ophyd/ophyd/userapi/scan_api.pyc in __call__(self, positioners, start, stop, npts, **kwargs)
    536         self.setup_scan(positioners, start, stop, npts, **kwargs)
    537         self.run(**kwargs)
--> 538 
    539     def setup_scan(self, positioners, start, stop, npts, **kwargs):
    540         """Setup scan along a regular path.

/home/xf23id1/Repos/ophyd/ophyd/userapi/scan_api.pyc in run(self, **kwargs)
    284         with self:
    285             self.check_paths()
--> 286             self.setup_detectors(self.detectors)
    287             self.setup_triggers(self.triggers)
    288 

/home/xf23id1/Repos/ophyd/ophyd/userapi/scan_api.pyc in check_paths(self)
    194                     raise ValueError('Scan moves positioner {} \
    195                                      out of limits {},{}'.format(
--> 196                                      pos.name, pos.low_limit, pos.high_limit))
    197 
    198     def __enter__(self):

AttributeError: 'numpy.float64' object has no attribute 'low_limit'
```

stuwilkins, 2015-02-18:
I think there is a problem with the stack traces…… this is not in setup_detectors.

I have seen this before with development deployments when the code is out of sync with the imported module.

S

same problem...

dchabot, 2015-02-18:
> 
> I think there is a problem with the stack traces…… this is not in setup_detectors. 
> 
> I have seen this before with development deployments when the code is out of sync with the imported module. 

Or a fetch without a pull...

ericdill, 2015-02-18:
Ah, yes... restarting ipython would help. Issue resolved. Thanks

closed 2015-02-18 by ericdill