bluesky/ophyd #74: BUG : Debug using print must go away

stuwilkins, 2015-02-24:
Looking through the current master branch there is a lot of debug using ``print()`` I think this should all be removed. We should use either the logging (after all that is what it is for!) or when a PR is created to master this should be stripped.

ericdill, 2015-02-24:
Are you talking about print statements in the `runengine`, the `userapi` or both?

stuwilkins, 2015-02-24:
All ….. I don’t think we should be using ``print()`` in production systems.

S

ericdill, 2015-02-24:
Are you suggesting that you dont want **any** print statements at all?  See this one for example: https://github.com/NSLS-II/ophyd/blob/master/ophyd/userapi/scan_api.py#L630

stuwilkins, 2015-02-24:
No as per the title … no debug prints. Statements to inform the user about what is going on (moving motors etc.) is OK.

S

gjwillms, 2015-03-17:
Can we discuss this further?  I'd like to see print() disappear, to be replaced by a logger.  The user should have the option to display minimal formatted information, e.g., [positioners] [user_detectors] [default_detectors] (time), at the INFO level and the current output would be provided at DEBUG.  As a terminal guy, there's way too much stuff getting dumped to my screen and I'd like to have the ability to control this, maybe even by providing additional formatting at custom logging levels.

tacaswell, 2015-03-17:
:+1: on proper use of logging

stuwilkins, 2015-03-17:
Agreed. The mechanism is in place, and we should use it and make sure it propagates up to the top layers. I agree more informative print is also needed on the beginning of a run…. scan_id etc.

Best wishes,
Stuart

gjwillms, 2015-03-18:
I'm going to work on this.  I'd apologize in advance, but I don't think I'll actually regret this action, or its consequences, and I hate to be dishonest.  Nevertheless, I'm sure it's a milestone of some sort...

closed 2015-04-09 by tacaswell