bluesky/ophyd #76: last_data is not what you think it is

ericdill, 2015-02-26:
`data` on this line https://github.com/NSLS-II/ophyd/blob/master/ophyd/userapi/scan_api.py#L307 is being returned as a dictionary of lists of value, timestamp lists.  the stats estimation is pretty badly broken because of this

stuwilkins, 2015-02-26:
This is fixed in the add_detectors branch and will be fixed on the next round.

S

Assigned #76<https://github.com/NSLS-II/ophyd/issues/76> to @stuwilkins<https://github.com/stuwilkins>.

stuwilkins, 2015-02-27:
P.S. This was not me that broke this, when we changed to tooples nobody checked the knockoff effects. Who assigned this to me anyway?

S
On Feb 26, 2015, at 4:54 PM, Wilkins, Stuart <swilkins@bnl.gov<mailto:swilkins@bnl.gov>> wrote:

This is fixed in the add_detectors branch and will be fixed on the next round.

S

dchabot, 2015-02-27:
> 
> P.S. This was not me that broke this, when we changed to tooples nobody checked the knockoff effects. Who assigned this to me anyway?

Didn’t realize this broke _after_ the value-timestamp tuples were introduced. Sorry.

ericdill, 2015-02-27:
I put it on your place since you are the only known consumer of last_data right now. Figured you would want to make changes to peak stats so that they worked like you intended

closed 2015-09-29 by danielballan