bluesky/ophyd #79: Potential key to start storing with user scans

ericdill, 2015-03-02:
It would be useful to start storing the data values that should be used to normalize detector columns during the ophyd scan.  Something like `normalize` would work.  I would assume that, for beamlines who need detector normalization, this should probably start getting dumped into the events automatically?

dchabot, 2015-03-02:
> It would be useful to start storing the data values that should be used to normalize detector columns during the ophyd scan. Something like normalize would work. I would assume that, for beamlines [...]

I’m probably missing something, but normalization is a post-acquisition activity.
I don’t think any reference to this belongs in ophyd.

stuwilkins, 2015-03-02:
No ….. I agree with Daron that all this “plotx, ploty, plotm” stuff doesn’t belong in the metadatastore or in the events. I like better his suggestion to use the olog for that. We could make a property which is this information which replay can pull (or even through data broker if needed).

The olog could easily be used for this. It also allows for updates.

S

stuwilkins, 2015-03-02:
P.S. If you look at CSX log entries most of the information is in there, we would just have to add the fields you want. What do you want?

S
On Mar 2, 2015, at 3:38 PM, Stuart B. Wilkins <swilkins@bnl.gov<mailto:swilkins@bnl.gov>> wrote:

No ….. I agree with Daron that all this “plotx, ploty, plotm” stuff doesn’t belong in the metadatastore or in the events. I like better his suggestion to use the olog for that. We could make a property which is this information which replay can pull (or even through data broker if needed).

The olog could easily be used for this. It also allows for updates.

S

danielballan, 2015-03-02:
Can someone bring me/us up to speed on how to put structures (programmatically readable) data into Olog? I might focus on this so that @ericdill can continue working on replay.

If we have a place to stash freeform useful data like this, we can built in user-friendliness (plot x, plot y, even silly things like plot style prefs) without poluting MDS. Since Olog entries are *editable*, putting ephemeral preferences in there seems OK.

arkilic, 2015-03-02:
If you are planning to add a lot of properties with bunch of attributes to Olog, you must avoid having too many of these free form entries. Olog performance decreases with increasing number of properties and attributes. If you don't care about speed and/or don't have too many properties. I can help you out with Olog

stuwilkins, 2015-03-02:
Hey Dan,

Take a look at:

https://github.com/NSLS-II-CSX/pyOlog

I made the SimpleOlogClient for exactly this purpose. Properties are just a dictionary.

S

klauer, 2016-03-03:
Ancient issue; no longer relevant

closed 2016-03-03 by klauer