bluesky/ophyd #83: Scan.detectors is not what a naive user thinks it is

gjwillms, 2015-03-17:
I think that there's a semantic problem with the detectors property in that it is intrinsically read-only.  A user will be likely to attempt to add a new detector with, e.g., Count.detectors.append([my_detector)], rather than adding it to the user_detectors list.  I offer up a few possible changes (in order of reversed personal preference):
1)  move this property name to _detectors so that a user has some warning that you don't intend to have this changed outside of the class methods,
2)  provide a setter that raises  "AttributeError", indicating that appending to this list is not supported, or
3)  provide a setter that transparently adds the provided detector list to the user_detectors list.

stuwilkins, 2015-03-17:
Hi Garth,

I agree that the user_detectors, default_detectors is very counterintuative. I am struggling for a good way to configure this? Any thoughts on what we can do to clean this all up?

Best wishes,
Stuart

gjwillms, 2015-03-17:
It's probably not helpful, but I'm actually glad to hear this.  I'm thinking about the issue.  If I'm honest, I'm tempted to say that we should eliminate the pair and return to just having a detector list.  I'm not sure that I have a detector that is so useful that I expect to find it in every scan I perform, so default_detectors is a little strange to me.  One of the things I may try tomorrow is creating a dict in the start-up files that represents a standard set of detectors for scans in each hutch and standard experiment configuration.  That, in conjunction with a helper function to set/unset detectors and summarize known configurations, might work.

closed 2015-11-06 by tacaswell