bluesky/ophyd #84: Put Stuart's directory-generating script into ophyd.utils

danielballan, 2015-03-17:


tacaswell, 2015-03-26:
and make it left-pad months/days with 0

tacaswell, 2015-11-06:
This should be done via the create-directory flag in AD

attn #165

tacaswell, 2015-11-06:
via @dchabot http://cars9.uchicago.edu/software/epics/NDPluginFile.html

klauer, 2015-11-06:
It may still be preferable to do it in Python, as you can set the permissions/ownership/etc as required. I don't think that exists in the AD file plugin, does it?

tacaswell, 2015-11-06:
The machine running blue sky may not be able to see the disks that ad is
writing to.

klauer, 2015-11-06:
Is this how other beamlines are setup (purposefully)? All of our machines at the HXN have access to where data is being written.

tacaswell, 2015-11-06:
see GPFS

klauer, 2015-11-06:
@tacaswell is that a reply to me? I don't know how GPFS is being setup.

danielballan, 2016-04-21:
Leaving this here -- will make a PR later:

```python
import os
import sys
import calendar

def makedirs(path):
    try:
        os.makedirs(path)
    except OSError:
        pass

def make_dir_tree(year):
    makedirs(str(year))
    for month in range(1, 13):
        print(month)
        makedirs(os.path.join(str(year), '%02d' % month))
        _, num_days = calendar.monthrange(year, month)
        for day in range(1, 1 + num_days):
            makedirs(os.path.join(str(year), '%02d' % month , '%02d' % day))

if __name__ == '__main__':
    year = int(sys.argv[1])
    make_dir_tree(year)
```

PR #323 (2016-05-16): ENH/TST: add makedirs with mode fix, make_dir_tree
Closes #84
closed 2016-05-16 by tacaswell