bluesky/ophyd #93: Timestamps, Processing and areaDetector

stuwilkins, 2015-03-22:
Ok, putting this as an issue so that we can discuss look at it. Attn. @dchabot @gjwillms @tacaswell .

After some discussion with Mark Rivers and playing with the AD Driver for the FastCCD I have managed to work out the issue with timestamps, and blocking. So, Firstly if you need to use the stats plugins (for example) you need to set them to blocking $(P)$(R)BlockingCallbacks. This will ensure that the put complete doesn't come back until the stats have processed.

Now in areaDetector 2.0 there is support for the epics records having the detector timestamp by setting the "TSE" field of those records. I tested this with the fastCCD (I take the timestamp of the first packet arriving at the detector electronics) and it looks good, even at 100Hz. Also all stats plugins have the same timestamp.

Also, the new HDF5 file plugin stores this info (timestamps) in the HDF5 file.

The info is here:

[areaDetector timestamps](http://cars.uchicago.edu/software/epics/areaDetectorTimeStampSupport.html)

gjwillms, 2015-03-27:
I don't see any downside to this.  From my point of view, this implements the intended behavior of the Stats plugin during a scan.  We don't have an immediate need for fast time-stamping, but it's great to have it in reserve.

stuwilkins, 2015-03-27:
Ok then i will implement it @tacaswell what is the state of my poor add_detectors branch?

stuwilkins, 2015-03-27:
P.s. I agree 100% this is actually what ad should do why it is not default is not clear to me.

Also, I noticed that AD 2.2 has a much simplified roi/stats plugin which should be much better for our intended use.

Best wishes,
Stuart

tacaswell, 2015-11-06:
This is currently working at CSX

attn #165

closed 2016-05-16 by klauer