bluesky/ophyd #96: Beamline control philosophy, ipython, and ophyd

gjwillms, 2015-03-26:
@stuwilkins @dchabot @klauer 
Gentlemen, I find myself in the midst of an existential crisis.  Normally, this would not affect you--unless you were unfortunate enough to find yourselves in my immediate vicinity; however, I was, rather unfortunately, thinking about ophyd during its onset.  As a result.  I have a *design question.*

I believe that there is merit in considering beam line control the problem of interfacing a small number of discrete systems:  e.g., the white beam slit, the horizontal offset mirror, the monochrometer.  These are, of course, comprised of many individual axes of control, but I believe that we do ourselves a disservice by attempting to control the beam line this way; doing so is micromanagement of the worst sort.  Said differently, we seem to be building, in ophyd, a rigidly functional approach to control on top of a flamboyantly object-oriented environment.  

As an example, suppose I wish to know the horizontal position and size of a slit.  In ophyd, I could accomplish this with "wh_pos([slt_pb_hcen,slt_pb_hsize])."  Would it not be better to encapsulate these properties of a slit in the object, calling them forth as "slt_pb.hpos.wh()"?

Opinions?

dchabot, 2015-03-26:
> On Mar 26, 2015, at 8:58 AM, gjwillms <notifications@github.com> wrote:
> 
> @stuwilkins @dchabot @klauer 
> Gentlemen, I find myself in the midst of an existential crisis. Normally, this would not affect you--unless you were unfortunate enough to find yourselves in my immediate vicinity; however, I was, rather unfortunately, thinking about ophyd during its onset. As a result. I have a design question.
> 
> I believe that there is merit in considering beam line control the problem of interfacing a small number of discrete systems: e.g., the white beam slit, the horizontal offset mirror, the monochrometer. These are, of course, comprised of many individual axes of control, but I believe that we do ourselves a disservice by attempting to control the beam line this way; doing so is micromanagement of the worst sort. Said differently, we seem to be building, in ophyd, a rigidly functional approach to control on top of a flamboyantly object-oriented environment.
> 
> As an example, suppose I wish to know the horizontal position and size of a slit. In ophyd, I could accomplish this with "wh_pos([slt_pb_hcen,slt_pb_hsize])." Would it not be better to encapsulate these properties of a slit in the object, calling them forth as "slt_pb.hpos.wh()"?
> 
> Opinions?

I think this is a bad idea.
In fact, I would argue your proposal is simultaneously “flamboyantly object-oriented” and “micromanagement of the worst sort”...

What is your objection to using, “slt_pb_hcen.position” ?
Is the problem that there are both procedural _and_ object-oriented api’s?

What you’re proposing lacks generality.
Perhaps I misunderstand what you’re trying to say, but are you proposing we provide mono, slit, mirror, etc classes and all their varieties too? 
That approach is counter to the idea of a framework, which we’re trying to provide.

There are two fundamental objects in ophyd that present an interface to hardware - Positioner and Signal/SignalGroup.
With a little wrangling, these could even be united to a SignalGroup with a lone signal. I’m not necessarily advocating that, though.

On top of the fundamental objects is a function-based, User API (wh_pos, mov, movr, etc).
These are intended to be used interactively, from the command-line, not programmatically.

gjwillms, 2015-03-26:
It is not my intention to offend.  

My observation is that if one adopts the paradigm implied by slt_pb_hcen.position, I need all of these: [slt_pb_hcen, slt_pb_vcen, slt_pb_hsize, slt_pb_vsize, slt_pb_inboardblade, slt_pb_outboardblade, slt_pb_topblade, slt_pb_bottomblade], to control a slit.  I have four of these on SRX.  That means that I have 32 postioners that I need to register as individual objects in ophyd just for my slits.  I think that you would agree that a wh_pos() would soon be unusable, and I would need to create specific lists of positioners to query.  This is the context in which I argue for a slit  'object' above.  When I'm doing everytihng else in a python shell, does it not make more sense to query slt_pb.hcen.wh() than wh_pos(slt_pb_hcen)?

I also think that one needs to be careful in adopting a strategy aspiring to complete generality, since cats are not dogs, let alone are they fish.  Ophyd currently designates everything as either a positioner or a signal.  This approach lacks generality(Is this dog a cat or a fish?).  My suggestion is not that ophyd provide classes for every device, but rather that it treats every device like an object, so that I am free to design a class that abides by an API.  In other words, a scan drives a method, not a PV.  Its readback is a method, not MOVN/DMOV.  For a simple positioner, the method and read-back are trival, but it allows for a more general device to be 'scanned.'  (This would have, for example, made XPD's recent problem a non-issue.)

In addition to the above, this approach would give me the ability to generate an object that completely describes, for example, a diode attached to a preamp.  A single object could be constructed that would contain readback, trigger, integration time, gain, etc.  I could then do the usual thing and scan a slit blade against the readback/trigger, or I could do more advanced things, for example, checking linearity by scanning the 'gain' against time and slit position (to change measured flux) in a mesh scan, with no obvious difference to the user at the keyboard.  That sort of flexibility seems desirable to me--but not easily implemented in ophyd--which led me to post the original comment.

dchabot, 2015-03-27:
> On Mar 26, 2015, at 1:19 PM, gjwillms <notifications@github.com> wrote:
> 
> It is not my intention to offend.

To be clear, it is not the ideas that are offensive, it’s the hyperbole and tone.

> My observation is that if one adopts the paradigm implied by slt_pb_hcen.position, I need all of these: [slt_pb_hcen, slt_pb_vcen, slt_pb_hsize, slt_pb_vsize, slt_pb_inboardblade, [...]

If you intend to move all 4 blades individually as well as coordinated gap & offset moves for each slit then, yes, configuration is required for all of those.
I don’t see how the quantity of channels and configuration would change with your model/example. 

>  I think that you would agree that a wh_pos() would soon be unusable, and I would need to create specific lists of positioners to query.

Creating beamilne-specific routines is one of the ideas we’re trying to enable. 

Would adding some filtering on (partial) strings help with the `wh_pos( )` usage?

```
def wh_mtr(fragment):
	mtrs = session_mgr.get_positioners()
	print [ v.position for k,v in mtrs.iteritems() if fragment in k]

wh_mtr(‘pb’)
```

> When I'm doing everytihng else in a python shell, does it not make more sense to query slt_pb.hcen.wh() than wh_pos(slt_pb_hcen)?

That is almost entirely subjective.

> I also think that one needs to be careful in adopting a strategy aspiring to complete generality, since cats are not dogs, let alone are they fish. 

???

> Ophyd currently designates everything as either a positioner or a signal. This approach lacks generality(Is this dog a cat or a fish?). 

Is having a minimal set of base objects that are broadly applicable not the definition of having generality?
Am I using the wrong words here?

> My suggestion is not that ophyd provide classes for every device, but rather that it treats every device like an object, so that I am free to design a class that abides by an API. In other words, a scan drives a method, not a PV.

This _is_ what is done now - scans use `Positioner.move_next( )`, and rely on asynchronous completion notifications.
I don’t believe the async reporting is a requirement though.

>  Its readback is a method, not MOVN/DMOV. For a simple positioner, the method and read-back are trival, but it allows for a more general device to be 'scanned.’ 

Have you looked at or been made aware of the PVPositioner class?
Use cases so far include hexapod, lakeshore temp controllers, and an Oxford cryostream...

Are you asking for the creation of a means of “moving” any device through a trajectory so that it can be a “Positioner” in a scan?
That is an idea i would entertain.

> (This would have, for example, made XPD's recent problem a non-issue.)

Fixing misbehaving hardware or server-side problems with client-side bandaids is poor practice.
The example you raise is a control system problem (epics), and should be fixed at that level.

> A single object could be constructed that would contain readback, trigger, integration time, gain, etc. I could then do the usual thing and scan a slit blade against the readback/trigger, or I could [...]

Flexibility is something to strive for and I’m willing to discuss how some of your ideas might be made to work, both in the future and with the current object model.
Some of your notions are at least partially addressed with the add_detectors branch. Concepts expressed there will make their way onto master, in one form or another.

If you have an actual alternative API or object model to propose, then please do that.

— dc

gjwillms, 2015-03-27:
If you find this conversation offensive.  I suggest that we suspend it and address the topics in a meeting.

tacaswell, 2015-03-27:
Please make sure I am invited to this meeting.

I think there is much less of a gap here that it appears.

dchabot, 2015-03-29:
I apologize for my overreaction.
Yes, we should meet to discuss.
Any time except for Tuesday is fine by me.

gjwillms, 2015-03-30:
I understand the pressure that you guys are under.  I'll try to find a better way to communicate with you.  I think that we should aim for later in the week, when everyone is more likely to be available.  This isn't nearly as urgent as many of the other things going on.

tacaswell, 2015-11-06:
Closing as I don't think there is a actionable issue here.

closed 2015-11-06 by tacaswell