bluesky/ophyd #100: Scan continues when limit switch is hit

skalbfleisch, 2015-04-03:
ascan does not stop or give an error when scanned motor hits a limit switch.

tacaswell, 2015-11-06:
This should probably be implemented as the `set` commands raising exceptions when limits are hit which _should_ bubble up to bluesky and kill the scan.

cowanml, 2015-11-06:
Seems like in most cases multiple things have already gone wrong if we
hit a hard limit?

I'd expect the scan to fail to start if an endpoint was past a soft limit,
suspend or stop if it somehow found itself passing a soft limit, and again
at a hard limit.  Is there any sort of detection/handling for stalled motors?

Ideally it should be made brutally obvious what the problem is and clear how
to recover and get back inside all limits.

PR #286 (2016-03-08): ENH: Positioner/EpicsMotor Additions Partially Fix Issue: #273
- [x] Homing
- [x] Calibration - set_current_position
- [x] Limit Switch Status
- [x] Velocity
- [x] Acceleration
- [x] Offset and Offset Freeze
- [x] Check the limit switches on the _move_changed prior to invoke _done_moving.
- [x] Tests

Closes #273 
Closes #100
closed 2016-03-08 by danielballan