bluesky/ophyd #101: Setting the acquisition time in scans

skalbfleisch, 2015-04-03:
ascan/dscan/other basic scans should allow to set the acquisition time through a command argument, e.g. “ascan(motor_mne, start_pos, end_pos, npts, acq_time)"

tacaswell, 2015-04-04:
What should this do if there is more than one detector attached to the scan?

skalbfleisch, 2015-04-06:
I can think of the following scenarios:

1. Detectors output TTL pulses as counts. The TTLs are counted by the VME-based scaler card. Then the count time of the scalar card should be set to the given acquisition time. The configured scalar channels are then read and stored. I think, this is already implemented.

2. Gated detectors: Before an acquisition is started, the detector needs to be put in gated mode by setting the correct PVs. The gate signal is then generated by the VME-based scaler card based on the given acquisition time. Before the next acquisition is started, it must be ensured that all data has been saved and the detector is ready.

3. Triggered detectors: Before an acquisition is started, the detector needs to be put in triggered mode and armed by setting the correct PVs. The acquisition time needs also to be set. The trigger signal is then generated by the VME-based scaler card. Before the next acquisition is started, it must be ensured that the acquisition is completed, all data has been saved, and the detector is ready.

4. Free run detectors: As free run detectors do not use any hardware-based trigger or gate signals, the acquisition is started and/or stopped on software-based signals, i.e., writing to PVs. Again, the given acquisition time needs to be set by writing it to the corresponding PV. The synchronization of this software-based approach to start/stop acquisitions is rather poor and therefore not preferred, but some detectors do not support hardware-based triggering/gating.

Most of the detectors are either read through the VME-based scalars or through areaDetector. This should then allow for some standardized way to interface the ophyd based scans to areaDetector. Certainly, some configuration of the detectors and the PVs is needed within ophyd.

If a detector is not supported in areaDetector, it is considered an exception and needs special treatment. Thus it is not part of this request.

klauer, 2015-07-17:
This was solved (well enough, at least) on the HXN beamline with `sync_dscan`. Marking this closed; feel free to re-open if it was decided this functionality should be pushed into ophyd itself.

closed 2015-07-17 by klauer