bluesky/ophyd #109: scans should die more gracefully with no detectors or positioners

tacaswell, 2015-04-09:
running `ct` with no detectors or positioners defined results in an exception from deep in the opyhd stack/metadatastore on a different thread which requires ctrl-c to kill.

dchabot, 2015-04-09:
Please provide a backtrace for this.

gjwillms, 2015-04-16:
I think that this is what Tom is referring to:
Ophyd [20]: ct.detectors
   Out[20]: []

Ophyd [21]: ct
----------> ct()
Stored '_scan_id' (int)
2015-04-16 12:55:12,530 [ophyd_session:INFO] Scan ID: 382
2015-04-16 12:55:12,530 [ophyd_session:INFO] Time: 2015-04-16T12:55:12.529094
2015-04-16 12:55:12,531 [ophyd_session:INFO] uid: 13509b88-010b-482c-beb8-f92e7eebf1cf
2015-04-16 12:55:12,531 [ophyd_session:INFO] Begin Run...
2015-04-16 12:55:12,531 [ophyd_session:INFO] 
2015-04-16 12:55:12,631 [ophyd_session:INFO] 
Exception in thread Scanner:
Traceback (most recent call last):
  File "/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-x86_64/egg/ophyd/runengine/runengine.py", line 234, in _start_scan
    data_keys=mds.format_data_keys(data_key_info))
  File "build/bdist.linux-x86_64/egg/metadatastore/commands.py", line 28, in inner
    return func(*args, **kwargs)
  File "build/bdist.linux-x86_64/egg/metadatastore/commands.py", line 286, in insert_event_descriptor
    event_descriptor.save(validate=True, write_concern={"w": 1})
  File "build/bdist.linux-x86_64/egg/mongoengine/document.py", line 224, in save
    self.validate(clean=clean)
  File "build/bdist.linux-x86_64/egg/mongoengine/base/document.py", line 323, in validate
    raise ValidationError(message, errors=errors)
ValidationError: ValidationError (EventDescriptor:None) (Field is required and cannot be empty: ['data_keys'])

<*ophyd hangs here until a sigint is sent*>

^C2015-04-16 12:55:39,207 [ophyd_session:INFO] End Run...
2015-04-16 12:55:39,207 [ophyd_session:INFO] 

Detector                     | Value
============================================================

dchabot, 2015-04-16:
Ok. Thanks.

@tacaswell: the scan should not die more gracefully - it should never be born.
I’ll submit a patch to the Scan class.

tacaswell, 2015-04-16:
Thanks, sorry for never following up on this.

commit 8c09a30c (2015-04-16): Merge pull request #117 from NSLS-II/scanapi_fixes

FIX : don't start scan if there are no detectors

fixes #109
closed 2015-04-16 by tacaswell