bluesky/ophyd #110: bug in dscan

ericdill, 2015-04-09:
```
Ophyd [32]: dscan(slt_ssa_ib, -.1, .1, 1, m3_xu_pos=m3_xu.get()[0])
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-32-1b957d6b3727> in <module>()
----> 1 dscan(slt_ssa_ib, -.1, .1, 1, m3_xu_pos=m3_xu.get()[0])

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in __call__(self, positioners, start, stop, npts, **kwargs)
    429             The number of intervals in the scan
    430         """
--> 431         self.setup_scan(positioners, start, stop, npts, **kwargs)
    432         self.run(**kwargs)
    433 

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in setup_scan(self, *args, **kwargs)
    505         current position and the scan range.
    506         """
--> 507         super(DScan, self).setup_scan(*args, **kwargs)
    508         self._start_positions = [p.position for p in self.positioners]
    509         self.paths = [np.array(path) + start

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in setup_scan(self, positioners, start, stop, npts, **kwargs)
    464 
    465         args = (positioners, start, stop, npts)
--> 466         self.scan_command = self._format_command_line(*args, **kwargs)
    467 
    468         # Calculate number of points from intervals

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in _format_command_line(self, *args, **kwargs)
    409     def _format_command_line(self, *args, **kwargs):
    410         """Return a string representation of the passed arguments"""
--> 411         cl_args = cmdline_to_str(*args, **kwargs)
    412         rtn = '{}()({})'.format(self.__class__.__name__, cl_args)
    413         return rtn

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in cmdline_to_str(*args, **kwargs)
    650         msg.append(', ')
    651         for key, value in kwargs.iteritems():
--> 652             msg.append('{}={}'.format(key, objects_to_str(value)))
    653 
    654     return ''.join(msg)

/nfs/xf05id1/mc/envs/ophyd_ng/lib/python2.7/site-packages/ophyd-0.0.7rc1-py2.7.egg/ophyd/userapi/scan_api.pyc in objects_to_str(iterables, iter_sep, iter_start, iter_stop)
    617         pass
    618 
--> 619     for i in iterables:
    620         # Is the object i iterable:
    621         if isinstance(i, tuple):

TypeError: 'float' object is not iterable
```

Note that:
```
Ophyd [34]: m3_xu.value
   Out[34]: [1.9106729999999998, 1.912, 'mm', 0, 1, 0]
```

dchabot, 2015-04-09:
> 
> Ophyd [32]: dscan(slt_ssa_ib, -.1, .1, 1, m3_xu_pos=m3_xu.get()[0])

What is this invocation trying to achieve?
A “snapshot” of m3_xu_pos at the start of the scan?

stuwilkins, 2015-04-10:
for a start should you not be using `m3_xu.position` if it is a motor? That should return a float not the list. 
Secondly, this is trying to make a string representation of the scan command as requested. It appears that when the kwargs were made freeform there are some pathological cases where this breaks. 

Please check with `m3_xu.position` and also m3_xu_pos=[m3_xu.position]

closed 2015-09-29 by danielballan