bluesky/ophyd #112: null AD path after scan

tacaswell, 2015-04-10:
While investigating https://github.com/NSLS-II/dataportal/issues/159 noticed bunch of files saved a long time after the first one.  

After a scan the AD plugin is left with that scans scan's path information.  We should probably null or reset this to what ever it was before so that user's do not accidentally write data into the data folders.

stuwilkins, 2015-04-10:
Tom,

This needs the enable / disable that we did yesterday with you and Dan. This should fix the problem

S

dchabot, 2015-04-10:
> While investigating NSLS-II/dataportal#159 noticed bunch of files saved a long time after the first one.
>
> After a scan the AD plugin is left with that scans scan's path information. We should [...]

Good catch.
That would be a nasty surprise…

tacaswell, 2015-04-10:
@stuwilkins To be clear, disabling the plugin by setting `self._write_plugin('EnableCallbacks', 0, self._file_plugin)` clears everything about the path?

If so, this is closed by #111 which is already merged and deployed to xf11id and xf28id.

stuwilkins, 2015-04-10:
That will disable the file writing plugin so that it ignores incoming images which will stop images from being written to disk.

S

tacaswell, 2015-04-11:
But if the user re-enables it for the CSS screen and then hits 'save' the old path will still be there?

stuwilkins, 2015-04-11:
Yes but with auto increment on it will not overwrite images. If you fiddle in CSS you get what you deserve however.

S

tacaswell, 2015-04-11:
We won't lose data, but there will be un-tracked files muddled up with tracked files which could be less than convenient for archiving etc.

stuwilkins, 2015-04-11:
True, but the user would have still had to re-enable the specific plugin with CSS. We just have to educate people not to do that. Just like many other aspects!

S

dchabot, 2015-04-11:
> 
> True, but the user would have still had to re-enable the specific plugin with CSS. We just have to educate people not to do that. Just like many other aspects!

I agree that a policy-based approach here is a reasonable first step.
But I also think there is ample room for improvement with the means and conventions used to organize data on-disk.

Organization by date presents at least one ambiguity.
eg: what happens when scans span the day and/or month rollover times?
If a scan is started on 04/11/15 at 11:50pm and finishes at 04/12/15 at 12:10am, are all data in the 04/11/ folder?

I realize that programmatically, retrieval complexities may be encapsulated, so much of this is hidden.
But, for a human being that may want or need to look through the filesystem (debugging?), there may be more useful organizational structures.

eg: organization by start-date _and_ run UID, or similar.

Just some thoughts...

stuwilkins, 2015-04-11:
>
> True, but the user would have still had to re-enable the specific plugin with CSS. We just have to educate people not to do that. Just like many other aspects!

I agree that a policy-based approach here is a reasonable first step.
But I also think there is ample room for improvement with the means and conventions used to organize data on-disk.

Organization by date presents at least one ambiguity.
eg: what happens when scans span the day and/or month rollover times?
If a scan is started on 04/11/15 at 11:50pm and finishes at 04/12/15 at 12:10am, are all data in the 04/11/ folder?

Currently the behavior is that the path is set at the call to configure() Therefore files do not span date directories but would exist in a single directory. I believe this is the best option.

I realize that programmatically, retrieval complexities may be encapsulated, so much of this is hidden.
But, for a human being that may want or need to look through the filesystem (debugging?), there may be more useful organizational structures.

eg: organization by start-date _and_ run UID, or similar.

Not a bad idea. Unfortunately this is a classic case of implementation by necessity before having this worked out. Perhaps this should be addressed before the next run.

Just some thoughts...

tacaswell, 2015-11-06:
attn #165

danielballan, 2016-01-14:
I think this is well addressed by unstaging.

To play devil's advocate for a moment: We could be more aggressive -- instead of returning the filepath and filename to their "original" values, we could set them to some gibberish placeholder that would require user attention.

But I think that might be going too far to protect users from themselves.

klauer, 2016-03-03:
@danielballan I believe the staging implementation addresses this sufficiently. Can this be closed?

closed 2016-03-03 by tacaswell