bluesky/ophyd #126: dscan - record total exposure time for the pe1 detector at XPD beamline

pavoljuhas, 2015-06-03:
The Perkin-Elmer pe1 detector is setup to sum up multiple exposure frames into a single tiff image that makes up one measurement.  The frame exposure time and number of frames are both configurable.  To my knowledge, there is no record of these values or the corresponding total exposure time in the databroker events as produced by dscan.

tacaswell, 2015-06-03:
This probably should be inserted into BeamlineConfig by the run engine.

cowanml, 2015-06-04:
beamline config?  do the other scan parameters go in beamline config?  That name is confusing.  Alot of people would expect 'beamline config' to be something infrequently changed, perhaps only by staff, unless there's automation or something.  Like default settings or what detector is installed or whatever changes are involved in changing techniques on a multi technique beamline.
RunConfig would be much better for how this seems to be used.

Do we only change what I think of as beamline config in yaml files or something?  But that info gets recorded along with scan params in BeamlineConfig?

tacaswell, 2015-06-04:
As currently understood by me, BeamlineConfig is where to put any
information which is not about the sample which does not change over the
course of the experiment which describes the configuration of the beam
line.  Things like detector exposure time seem like the fall squarely into
that category, as does things like sample-to-detector distance, the model
of the goniometer used, and all fe the calibration values (even if we are
not capturing any of that right now).  If you are changing the exposure on
a per-event basis then it needs to be something that goes into your event
streams.

cowanml, 2015-06-04:
I think I understand how it's used, and it functions fine, just find the name throws me off.
I guess I'm really used to dividing things up into:

   ~ changes often, with every run/scan, customized params  (exposure time)
     ... data collection parameters, level 0 data,
     write once, never change

   ~ changes rarely, and/or only by staff (gonio, det, and default
     params [including exposure time])... beamline configuration, level -1 settings/state,
     could be overwritten as long as values are copied into mds when actually used or saved
     in the archiver

John and I are trying to find the best place to stick various bits of
beamline state info and defaults, and maybe some persistent per user settings.
BeamlineConfig sounds right, but isn't quite.

-matt

arkilic, 2015-06-04:
Beamline's configuration should not be managed based on information within this collection. As @tacaswell  also suggested, this is just a simple way for analysis to have some information regarding the configuration that was part of the process which resulted in the given run data. Managing beamline configuration with lots of updates and user interaction requires more logic than a simple collection like BeamlineConfig. 

If you're thinking of managing your beamline configuration with lots of writes/updates, there are other tools, such as RocksDB that you might want to look into. 

John and I discussed his specific application and he suggested his confusion was resolved. His use-case is different and I believe he has already decided to stick to his existing methods. We named it BeamlineConfig beacuse naming it BeamlineRunConfigForGivenRunHeadersAtTheSpecificRunTimeForAnalysisPurposes was plausible :)

tacaswell, 2015-06-04:
@ericdill has a PR in to allow corrections to be over-laid on top of MDS is a safe way https://github.com/NSLS-II/metadatastore/pull/150

The `RunStart` documents are flexible so if you want to split your data into two (`BeamLineConfig` and `BaseConfig` or `BeamlineConfig` and `RunConfig`?) I think that is ok.  We have a fair amount of flexibility on these documents still as ohpyd has not actually been emitting them with much meaningful content and we have no consumers relying on a given spec.

I sort of suspect that the exact layout of these documents is going to end up being beam line specific anyway.  We probably should tack a `BLC_schema` string into runstart as well.

tacaswell, 2015-11-06:
attn #165 

Need to shove configuration into the output of `describe`.  The machinery is already there in bluesky to handle this.

closed 2016-03-03 by klauer