bluesky/ophyd #135: Extract all useful info from epics records and put it in MDS

danielballan, 2015-07-02:
That is, add extra fields to the data keys -- which are dynamic, so this not a breaking change --such as datatype, units, precision (and/or a format string), and so on.

cowanml, 2015-07-02:
On Wed, 1 Jul 2015, Dan Allan wrote:
> 
> That is, add extra fields to the data keys -- which are dynamic, so this not a breaking change
> --such as datatype, units, precision (and/or a format string), and so on.

Just want to make sure we're not inadvertently duplicating
functionality the epics archiver provides?

-matt

danielballan, 2015-07-02:
I think the point of MDS is that it does duplicate a lot of what the epics
archived provides, but it collects data faster than 10Hz and includes more
user-given metadata and context.

Right?

cowanml, 2015-07-02:
If mds is acting as an epics client pulling info out of an ioc, it's
not able to do anything the archiver couldn't?

mds is the place to correlate pv data with other pv data and/or the
metadata it stores.  Duplicating some pv data isn't inherently bad, I
just saw "Extract all useful info from epics records" and wanted to
make sure we weren't inadvertantly duplicating effort.

danielballan, 2015-07-02:
Is there code in channelarchiver that reads units, datatype, etc. out of an epics record?

dchabot, 2015-07-02:
> 
> Is there code in channelarchiver that reads units, datatype, etc. out of an epics record?

This exists natively in EPICS.
We are simply not performing every get/put/monitor as type DBR_CTRL in ophyd.

dchabot@xf23id1-srv1--> caget -d DBR_CTRL_DOUBLE XF:23ID1-ES{Dif-Ax:Th}Mtr.RBV
XF:23ID1-ES{Dif-Ax:Th}Mtr.RBV
   Native data type: DBF_DOUBLE
   Request type:     DBR_CTRL_DOUBLE
   Element count:    1
   Value:            0.00114573
   Status:           NO_ALARM
   Severity:         NO_ALARM
   Units:            deg
   Precision:        3
   Lo disp limit:    -3304.64
   Hi disp limit:    3895.36
   Lo alarm limit:   nan
   Lo warn limit:    nan
   Hi warn limit:    nan
   Hi alarm limit:   nan
   Lo ctrl limit:    -3304.64
   Hi ctrl limit:    3895.36

danielballan, 2015-07-02:
Right. It looks like [pyepics exposes this](http://cars9.uchicago.edu/software/python/pyepics3/pv.html#pv.precision). I'm suggesting that we capture precision and units in the EventDescriptor Documents so that, among other things, we can avoid printing 10 digits of precision to the screen when only 3 digits are meaningful.

danielballan, 2015-09-29:
@dchabot This feature request keeps coming up. It would be nice if `.describe()` included `unit` and `precision`. Do you have the time to look at this?

dchabot, 2015-09-29:
Yes, I'll put a PR together.

klauer, 2015-09-30:
One thing @dchabot and I found yesterday was that `epics.PV().units` returns a byte string in 3.x. Need to get this fixed upstream in pyepics.

klauer, 2016-03-03:
Fixes upstream in pyepics were merged, and closed by #225

closed 2016-03-03 by klauer