bluesky/ophyd #136: Do not accept 'None' as a number

danielballan, 2015-07-02:
Raise or at least re-read when the ioc returns `None` for a value that is expected to be numeric.

Real-life example:

```
In [17]: theta.read()
Out[17]: {'theta': {'timestamp': None, 'value': None}}
```

klauer, 2015-07-17:
To clarify, EPICS IOCs don't have a concept of ```None``` - this is unfortunately how pyepics indicates an error/timeout. `epics.caget` works the same. For example:

```python
In [1]: pv = epics.PV('_INVALID_PV_NAME_')
In [2]: value = pv.get()
In [3]: print value
None
```

danielballan, 2015-07-17:
Right, this could be addressed at the pyepics level instead. For now, it seems easier to catch it in ophyd and raise an exception instead of saying "The motor position is None."

tacaswell, 2015-07-20:
:+1: on ophyd raising on a failed read.

tacaswell, 2015-11-06:
@dchabot Should this be marked as done due to #144 ?

dchabot, 2015-11-06:
Yes, #144 covers this.

closed 2015-11-06 by dchabot