bluesky/ophyd #138: user_detectors and default_detectors

ericdill, 2015-08-05:
We should convert `user_detectors` and `default_detectors` to be a set, not a list. There is no use case where you want to add the same python representation of a detector twice. In fact this might be what is causing the strange bug at CHX where N files are written to disk, but the file index in filestore is 2n+1 where 0 < n < N

ericdill, 2015-08-05:
cc  @danielballan @tacaswell @afluerasu @yugangzhang

klauer, 2015-08-06:
I'd agree if you mean using an `OrderedSet` - sometimes it's useful and/or necessary to trigger the detectors in a certain order.

Even so, this only would help when re-using the same Python object. If the user makes two detector instances controlling the same PV, the `set` wouldn't help as the object hashes would differ.

stuwilkins, 2015-08-06:
I agree with Ken, this is not so easy as it is very difficult to define uniqueness on an object. Is this a situation where we want to think about hashing?

Best wishes,
Stuart

tacaswell, 2015-08-06:
Yes, python objects can define a `__hash__` and `__eq__` methods.  This should probably be based on the PV root or something similar.

In the case where the detector trigger order matters you probably should not be using the canned 'magic' scanning functions anyway.

The other place we should add a check is that when creating events we never clobber an existing key.

klauer, 2015-08-06:
Whether the user should be using "canned 'magic' scanning functions" aside, I disagree about the ordering. In the context of synchronous tasks, when would indeterminate ordering ever be a good thing?

tacaswell, 2015-08-06:
What is an example where detectors need to be triggered in the right order in a step scan?

These global data structures are only used by the spec-like interface, not generally, which are only used for step scans.  

In a step scan the assumption on the analysis side is that everything bundled in an event document are 'at the same time' so the order of triggering can not matter from that point of view.  Exposing an implementation detail of the collection that can not make it through to the analysis side just adds more complication.

If triggering one detector has side effects that affect another detector then that set of interdependent detectors should probably be bundled into a single python object with trigger/read methods that is independent of any other detector.  

If there are detectors that _must_ be triggered/read in a specific order that breaks the core assumption of the `Event` documents and should be bundled into separate event streams which is not something that can be done with the spec-like api right now.

klauer, 2015-08-06:
Are we talking about the BlueSky implementation (of which I have no clue and as such am not referring to in this comment) or the current scan engine implementation?

Here's a pretty simple example. I want a detector to be gated on the hardware level -- acquiring at the exact same time (well, on the order of nanoseconds) for one point in a step scan.

* Tell the scaler to count and it generates the gate/trigger pulse
* Scaler is used as the "master"
* Other detector is the "slave", waiting for this pulse

So, you first need to arm the slave detector to wait for this trigger pulse, then you acquire with the scaler to send the trigger pulse. If the scaler is triggered prior to the slave detector, it will likely miss the pulse. The trigger order then matters. You could then add as many slave detectors in the same spot as necessary (or none at all).

Sure, these could be bundled in a different detector object with its own separate configuration and such, but you then don't have the simple `ascan.detectors` one-liner to configure everything you want triggered.

It's not a _huge_ deal either way as there are alternative solutions, but I have to admit I'm a little surprised at your being so vehemently in favor of random ordering...

stuwilkins, 2015-08-06:
I agree with Ken on this, do you think we should schedule a 30 or 40 mins discussion on this … this is quite an important aspect.

S

danielballan, 2015-08-06:
By insisting that ordering is not allowed to matter -- by insisting that anything in the global detectors list must be non-interfering -- you put the burden on the detector object and not the logic of the scan. You can still use `ascan.detectors`. You just have to make a `SlaveDetector` and a `MasterDetector` with custom `trigger` methods. Such a software design is a better reflection of the underlying hardware: that is, these detectors are special.

The scan logic can get complex -- meshes, winding ("snaked") trajectories, etc. -- and all of that is simpler if you know that detector objects are taking care of themselves.

dchabot, 2015-08-06:
> 
> By insisting that ordering is not allowed to matter -- by insisting that anything in the global detectors list must be non-interfering -- you put the burden on the detector object and not the logic of the scan.

IMO, @danielballan/@tacaswell, this is a mistake.
The device objects in ophyd are intended to reflect a notion of device drivers - they implement mechanism, not policy or logic.

I appreciate that it gets the job done, but the policy implemented for dark-frame acquisition make me cringe...

> You can still use ascan.detectors. You just have to make a SlaveDetector and a MasterDetector with custom trigger methods. Such a software design is a better reflection of the underlying hardware: that is, these detectors are special.

No, the detectors are not special. The policy (logic) of a use-case is where the “speciality” is introduced.
i.e. in the scan logic.

> The scan logic can get complex -- meshes, winding ("snaked") trajectories, etc. -- and all of that is simpler if you know that detector objects are taking care of themselves.

That’s actually a good example of my point: the devices know nothing of how they’re being used to generate a trajectory (policy), they only know how to respond to a “move”.

Just my $0.02.

danielballan, 2015-08-06:
> That’s actually a good example of my point: the devices know nothing of how they’re being used to generate a trajectory (policy), they only know how to respond to a “move”

I will take some time to think about this point. We seem to look at the same issue and draw opposite conclusions.

tacaswell, 2015-08-06:
On the other hand we already have an idea of pseudo motor which abstracts away a collection of motors to move along a synthetic axis in a way that is transparent to the user, that can be generalized to detectors as well.  Ex

```python
class BunchDetectors:
    def __init__(self, leader_detector, follower_detectors):
        self._followers = follower_detectors
        self._leader = leader_detector 

    def trigger(self):
        statuses = []
        for fd in self._followers:
            st = fd.arm()
            statuses.append(st)
        st = self._leader.trigger()
        statuses.append(st)
        return merged_status(statuses)  # this will be subtle, but doable
   
   def read(self):
        out = self._leader.read()
        for fd in self._followers:
            out.update(fd.read())
        return out

    def describe(self):
        return merged_describes
```

which nicely encapsulates the logic of linkage between the detectors that need to be coupled.  

This also keeps the meaning of `trigger` uniform.  In your example, `trigger` on some objects means 'arm and wait for a pulse' and on others means 'send the signal to start collecting data'.  Implemented this way it would mean that if a user removes the leader detector and only triggers the follower detectors the scan would hang as the gate pulse would never be generated.

It is not random order per say, just not guaranteed order and avoiding side-effects.

danielballan, 2015-08-06:
I guess the question of where "policy" belongs depends on whether you consider triggering hardware to be a part of the experiment design (it belongs on the scan!) or an implementation detail (it belongs in the object). To put it another way, it depends on whether you'd rather write a zoo of scans or a zoo of objects.

We don't need to take a position on that to resolve this issue. And maybe we never should take a position, instead leaving it to developer preference.

In an offline conversation, @ericdill and I became convinced that the detector order should be deterministic (i.e., a list). If nothing else, detectors should not be "configured" and "deconfigured" in randomized order. We can talk about triggering order in a future discussion.

To prohibit accidental duplication, we should validate the detector list by calling `.describe()` on its elements and checking that each data key is unique. That's the uniqueness that matters, not the Python object id.

tacaswell, 2015-11-06:
Closing this as this is a bluesky concern now.

closed 2015-11-06 by tacaswell