bluesky/ophyd #154: add time vector to events which contain time series measurements

tacaswell, 2015-10-04:
Some data collection methods involve taking time series at each position of a step scan.  Analysis may need to know the exact times each of the sub-measurements  we taken.  A proposed approach is to add a vector of timesptamps to the event which is lock-step with the time series.

This means we need to relax the constraint of 'scalar only' directly in MDS.  This seems to be less of a can of worms than letting the BS session start writing out files to disk and easier than forcing every IOC that might produce vector data to be able to handle file writing/integration with FS.

@stuwilkins

dchabot, 2015-10-05:
@tacaswell: at least myself and @arkilic would like to review this proposal in more detail. Maybe @cowanml too?

Can we meet to discuss v soon? Today or tomorrow?

tacaswell, 2015-10-05:
Tomorrow is better

arkilic, 2015-10-05:
@dchabot @tacaswell I am free all day tomorrow

tacaswell, 2016-10-19:
We have relaxed the 'scalar only' constraint by putting in MCA data and this is effectively happening at ISS with the pizza box.

closed 2016-10-19 by tacaswell