bluesky/ophyd #181: Review devices and make sensible defaults for read_fields, configure_fields

danielballan, 2015-12-16:
Can you look at this @dchabot? I think @klauer took a first pass but it's probably worth collecting more than one informed opinion here.

klauer, 2015-12-16:
I made split-second decisions on these. They almost certainly should be thought about more thoroughly and fixed up.

danielballan, 2016-01-11:
EpicsScaler: Remove `preset_time` as a read field and add `time` (will be `scaler_time`)
PVPositioner: Add `readback` and `setpoint` conditional on them not being None.

dchabot, 2016-01-11:
Sorry, this fell off my radar.
I'd like to see consistent naming used for fields that represent 'exposure_time', 'integration_time', etc.
Thoughts?

klauer, 2016-01-11:
@dchabot :+1:

danielballan, 2016-01-11:
So, to be clear, consistent between devices at the expense of being consistent with EPICS record naming? I think that's the right choice.

dchabot, 2016-01-11:
yes, exactly. well said.

danielballan, 2016-01-11:
@dchabot I leave this to you, the one who knows best what these names actually mean. I'm going to tag v0.2.0 as planned today and start deploying, knowing that we can change a few names in v0.2.1. Pretty minor compared to the mess we are about to make....

klauer, 2016-05-16:
`count_time` was standardized upon in the new bluesky/global_state setup. `Device.stage()` should read this value and do with it what it pleases.

Regarding `configuration_attrs` and `read_attrs`, we have been tweaking these as time goes on on a per-device basis. Though they could use a thorough review, in the end this is up to the beamline staff to decide what they want stored.

closed 2016-05-16 by klauer