bluesky/ophyd #182: Expose ArrayData in area detector objects

danielballan, 2015-12-16:
It won't normally be included in `read_attrs` or `configuration_attrs`, but it would be convenient to be able to access the data for debugging purposes. It's a little tricky to fit `image1:ArrayData1` into the existing class hierarchy, so we'll wait for @klauer.

klauer, 2016-01-05:
What do you mean by 'be able to access the data'? At what point?

I believe `ArrayData` itself is already available, assuming you create the image plugin component on your detector (ref: [ArrayData](https://github.com/NSLS-II/ophyd/blob/master/ophyd/controls/areadetector/plugins.py#L141))

danielballan, 2016-01-05:
By "access the data" I just mean `det.image1.array_data`. I don't think that attribute is currently implemented, but I could be wrong.

klauer, 2016-01-05:
In that case, check my reference above - it exists and should (hopefully) be working.

danielballan, 2016-01-11:
The source of confusion here was that area detector's don't have an image plugin by default. That has to be specified in a custom subclass (probably in the beamline's IPython profile).

closed 2016-01-11 by danielballan