bluesky/ophyd #198: Accessing array_data produces a segfault

danielballan, 2016-01-11:
I have not yet tried to reproduce this on the test stand, but at least at CSX this happens:

```python

In [1]: import ophyd.areadetector.trigger_mixins as tr

In [2]: import ophyd

In [3]: class MyDet(tr.SingleTrigger, ophyd.areadetector.AreaDetector):
    image = ophyd.Component(ophyd.areadetector.ImagePlugin, 'image1:')
   ...:

In [4]: det = MyDet('XF:23ID1-ES{Dif-Cam:Beam}', name='dif_beam_cam')

In [5]: det.stage()

In [6]: det.trigger()
Out[6]: <ophyd.ophydobj.DeviceStatus at 0x7f1f0ca300b8>

In [7]: _.done
Out[7]: True

In [8]: det.image.array_data
Segmentation fault
```

Other attributes of `image`, such as `array_size`, work properly.

stuwilkins, 2016-01-11:
I may have seen this before. Is the environmental variable EPICS_CA_MAX_ARRAY_BYTES set? I remember with an older version of pyepics I saw that problem.

I have:

EPICS_CA_MAX_ARRAY_BYTES=10000000

danielballan, 2016-01-11:
@stuwilkins Thanks for the suggestion but I already have that in place. Any other ideas?

```
$ env | grep EPICS
EPICS_CA_AUTO_ADDR_LIST=NO
EPICS_CA_ADDR_LIST=10.23.0.255
EPICS_CA_MAX_ARRAY_BYTES=10000000
```

stuwilkins, 2016-01-11:
Then no …. the only other thing I remember is you need to specify not to monitor these large PVs. Apart from that I have no idea.

S

danielballan, 2016-01-12:
Resolved by upgrade to pyepics 3.2.5rc3

Related issue: https://github.com/NSLS-II/ophyd/issues/121

closed 2016-01-12 by danielballan