bluesky/ophyd #202: Signals should implement read_configuration

danielballan, 2016-01-12:
Discovered while trying to add `em.channels.chan22` as a detector. It implements `read` and `describe`; it should also implement `read_configuration` so it can be used by bluesky.

klauer, 2016-01-12:
When would we ever want `read_configuration` on a simple thing like a signal? Isn't this something bluesky should consider optional?

danielballan, 2016-01-12:
If you don't think it's ever useful on a signal, then I'm happy to make it optional. But would it be useful to eventually capture things like limits in signal's configuration?

klauer, 2016-01-12:
Good point. @dchabot had mentioned grabbing all that sort of information when pyepics successfully connected.

danielballan, 2016-01-12:
In that case, see #203 :- )

PR #203 (2016-01-12): Make signal implement all required bluesky methods
Closes #202
closed 2016-01-12 by tacaswell