bluesky/ophyd #213: `read_attrs` only addresses keys which map cleanly to sub-devices

tacaswell, 2016-01-14:
ex, for an AD object

```python
sd.read_attrs = ['tiff1']
sd.tiff1.read_attrs = []
```

Is what is required to get _just_ the `image` key out of `sd.read()` not _everything_ AD knows about.

tacaswell, 2016-01-14:
I think this would be clearer if we renamed `read_attrs` -> `read_cmpts`, that is the list of a `Devices` components that it will consult as part of the read process.

Tuning how a `Devices` or  `Signal` tunes what keys it includes in the read dict that it it's self produces is a different problem.

klauer, 2016-01-14:
It's certainly too much information in the case of plugins, but isn't it useful to be able to say "I want `subdevice.**"` to be read? That's why it's implemented as-is.

klauer, 2016-01-14:
`read_cmpts` then could potentially include components that aren't even part of the `Device` hierarchy. `__getattr__` stuff probably could be removed in that case, though.

tacaswell, 2016-01-14:
I am super confused

klauer, 2016-01-14:
Am I misunderstanding your suggestion?
```python
device.read_cmpts = [device.cpt1]
```

tacaswell, 2016-01-14:
I am suggesting to leave the functionality identical, but to rename it because it confused me.

I was mapping attrs -> keys that will be in the dictionary returned by read, instead a list of the name of sub-components of the current device that will be consulted as part of the read.

klauer, 2016-01-14:
It was originally `signals` then renamed to `attrs` because of the exact problem you're describing. `attrs` is at least what Python would call them.

danielballan, 2016-01-15:
To sum up:

* "read_signals" is problematic because the names can refer to Signals or Devices.
* "read_attrs" is problematic because not all objects attributes qualify (only those that support `.get()`).
* "read_cmpts" is heavily abbreviated for my taste (I can hear myself spelling out: "No, C-M-P-T-S....")
* All of these might suggest that their contents are objects, not their string names.

How about:

* `signal_names` -> `component_names`
* `read_signals` -> `read_names` and likewise for `configuration_*`

Better make these changes Monday if we make them at all. Sanjit and Milinda already took notes on the existing terminology.

klauer, 2016-02-18:
@tacaswell  @danielballan  Are you ok with closing this? I think we agreed that all of the proposed names have their own issues, so we may as well stick with what we have (and it being somewhat late to rename things and all...)

danielballan, 2016-02-18:
Yeah, I came to the same conclusion. If @tacaswell is OK,  close it.

closed 2016-05-16 by klauer