bluesky/ophyd #220: PVPositioner may not marking its status object as 'done' correctly.

danielballan, 2016-01-15:
The temperature controller (PVPositioner) at XPD reaches 'Done' according to EPICS:

```
The xf28id1@xf28id1-ws2:~$ caget XF:28IDC-ES:1{Env:01}Cmd-Busy
XF:28IDC-ES:1{Env:01}Cmd-Busy  Busy
xf28id1@xf28id1-ws2:~$ caget XF:28IDC-ES:1{Env:01}Cmd-Busy
XF:28IDC-ES:1{Env:01}Cmd-Busy  Done
```

But bluesky is still waiting indefinitely (for at least 10 minutes):
```python
RE.state: running
msg: set: (CS700TemperatureController(prefix='XF:28IDC-ES:1{Env:01}', name='cs700', egu='', timeout=0.0, read_attrs=['setpoint', 'readback', 'done', 'stop_signal'], configuration_attrs=[], monitor_attrs=[], settle_time=0.05, limits=None)), (298.5,), {}
  response: MoveStatus(done=False, elapsed=0.0, success=False)
About to process: <bound method RunEngine._wait of <bluesky.run_engine.RunEngine object at 0x7fad808f64e0>>, wait: (None), ('A',), {}
```

This is the definition of the object in question. We are on ophyd master.

```python
lass CS700TemperatureController(PVPositioner):
    setpoint = C(EpicsSignal, 'T-SP')
    readback = C(EpicsSignalRO, 'T-I')
    done = C(EpicsSignalRO, 'Cmd-Busy')
    stop_signal = C(EpicsSignal, 'Cmd-Cmd')

    def trigger(self):
        # There is nothing to do. Just report that we are done.
        # Note: This really should not necessary to do --
        # future changes to PVPositioner may obviate this code.
        status = DeviceStatus()
        status._finished()
        return status
```

I believe this is only tangentially related to #219.

klauer, 2016-01-15:
What's `done_value` set to and what should it be?

danielballan, 2016-01-15:
While `caget` shows 'Busy', `cs700.done.get()` returns `1`. And `cs700.done_value` is `1`. From this I gather that `cs700.done_value` should be `0`.

cc @sghose

klauer, 2016-01-15:
In that case, I believe it should complete as expected if you set `done_value` to 0.

danielballan, 2016-02-18:
This did not immediate fix the entire issue, but it seems that the rest was hardware trouble. It has been working for weeks, so I'll close. Reopen if anyone has more information about what was going wrong.

closed 2016-02-18 by danielballan