bluesky/ophyd #230: Conclusions from preliminary discussion about positioners

danielballan, 2016-01-21:
PVPositioner -> something with a word like "Template" in it
PseudoPositioner -> something with a word like "Template" in it

Refactor `move` in Positioner to express a "synchronous" move in terms of a "async" move, merely watching the status object.

Add docstrings to PseudoPositioner and more docstring to Positioner.

Remove "soft positioner" args, attributes, and assumptions from Positioner base class.

Add an `actuate` method to Positioner and make basically the ultimate source of both `set` and `actuate`. Obviously need to add this Msg to bluesky plans.

* get and put -- low level, comprehensive
* Positioners set(*args, **kwargs), actuate() -- pick apart setting the set point and moving to the set point
* All devices trigger() and read() -- analogue to set and actuate. Trigger has nothing to do with moving, setting.
* Maybe `move` is refactored as set + actuate (+ wait for status to complete) to maintain current actual behavior

klauer, 2016-02-22:
Revisit naming of `Template`s - no one was happy about `PVPositionerTemplate` `PVPositionerPCTemplate`. Will leave for next refactor phase.

tacaswell, 2016-10-19:
Closing this as no longer useful.

closed 2016-10-19 by tacaswell