bluesky/ophyd #245: CHX wants some way to disable file-saving interactively

danielballan, 2016-01-27:
CHX wants some way to disable file-saving interactively. Sometimes they only care about ROI sums from the stats plugin, and they want to conserve disk space by not writing TIFFs of the full image all the time.

I see two options:

1. Instead enabling the plugin as part of staging, instead *check* whether it is enabled and short-circuit the rest of staging / all filestore interaction if it is not.
2. Add a new boolean attribute, separate from EPICS, that can be flipped on or off.

Maybe there are other good options I haven't thought of. I defer completely to @klauer and @dchabot. Tell me what makes sense and I will implement it.

klauer, 2016-01-27:
ROIs can update as they please, but we choose whether or not to save them by `read_attrs`.

Shouldn't we have similar control over what gets staged (and thus saved to disk/read/linked to filestore) by `FilePlugins` or other sub-devices?

dchabot, 2016-01-27:
I need to devote some time to consider this. Of your options, I would lean towards 2, with some reservations. This is something that definitely should be hashed out in person...

tacaswell, 2016-01-27:
I like number 2.  There already is some similar logic at XPD using a soft signal to configure the tiff-squashing.

klauer, 2016-01-28:
So `read_attrs` currently white-lists what we want to record. The argument against requiring a perfectly configured white-listed set of `stage_attrs` is compelling. 

However, what if we could selectively black-list plugins from being staged? (Just throwing this out there for discussion, in any case, I'm not saying it's the right solution)

danielballan, 2016-03-11:
Currently, if a plugin is exposed to Python it is always enabled. (If disabled, it is enabled upon `stage()` and *not* disabled by `unstage()`.) Maybe this was a bad choice. If we let users disable the TIFF plugin, we don't need to do anything special.

In fact this auto-enabling has caused confusion at both SRX and CHX. Remove it?

danielballan, 2017-08-02:
We will address this as part of #412.

tacaswell, 2017-10-23:
Closed via #326

closed 2017-10-23 by tacaswell