bluesky/ophyd #246: need a way to make sure AD does not burn all the CPU monitoring diagnostic cams

tacaswell, 2016-01-27:
With Diag6 spinning at CSX ipython burns 102% of a cpu in the background, with the cam stopped it uses ~5%.   With it spinning the ipython prompt is laggy and super annoying to use.

@klauer @dchabot

klauer, 2016-01-27:
Perhaps all `EpicsSignal`s on large composite devices (such as AreaDetector cams, plugins and such) should not only be lazy by default but also `auto_monitor=False`.

`ArrayData` on the image plugin looks like it is missing this and is a very likely culprit for eating too much CPU.

dchabot, 2016-01-27:
I agree with @klauer's assessment. Try passing `auto_monitor=False` to the `Component` definition of `ArrayData` and re-run the test.

NOTE - there are likely additional candidate Components that would benefit(?) from this treatment.

PR #249 (2016-01-28): FIX: re-initialize automonitor when an EpicsSignal subscription is made
Replaces #248 
Should fix #246 

Tests passing locally, but we should make extra sure the logic is correct for re-initialization of `epics.PV`s
closed 2016-01-28 by dchabot