bluesky/ophyd #254: Hardcoded "cam1" at ophyd detectors

hhslepicka, 2016-02-01:
By default the code assumes that everyone uses "cam1" in their area detector configuration. At LIX and some other beam lines that I checked the value for this parameter is empty. Is it possible to make it configurable in some way?

Here is the point where "cam1" is hardcoded.
https://github.com/NSLS-II/ophyd/blob/adacf52896c84c52e77757bc83e55771edecf2e4/ophyd/areadetector/detectors.py#L147

In order to get something running I did the following MacGyver:
```python
class LixProsilicaDetector(DetectorBase):
    _html_docs = ['prosilicaDoc.html']
    cam = Cpt(cam.ProsilicaDetectorCam, '')
```

Thanks!

klauer, 2016-02-01:
You can subclass and override (shadow) `ProsilicaDetector`'s `cam` component with the appropriate suffix. Implementing it as you describe is just fine, too.

hhslepicka, 2016-02-01:
:+1: No problem! Closing the issue!

closed 2016-02-01 by hhslepicka