bluesky/ophyd #256: wh_pos breaks with a disconnected device

hhslepicka, 2016-02-03:
When running the wh_pos() command with a disconnected motor on the list the following error appears.

My ophyd version is: 
```python
In [7]: ophyd.__version__
Out[7]: '0.2.0rc2+14.g89cc175'
```

Here is the stack trace:
```python
In [5]: wh_pos()
---------------------------------------------------------------------------
DisconnectedError                         Traceback (most recent call last)
<ipython-input-5-85b143a9e523> in <module>()
----> 1 wh_pos()

/opt/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in wh_pos(positioners)
    413     >>>wh_pos([m1, m2, m3])
    414     """
--> 415     positioners = _normalize_positioners(positioners)
    416     _print_pos(positioners, file=sys.stdout)
    417

/opt/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in _normalize_positioners(positioners)
    110     if positioners is None:
    111         # Grab IPython namespace, recursively find Positioners.
--> 112         res = get_all_positioners()
    113     elif isinstance(positioners, (Device, Positioner)):
    114         # Explore children in case this is a composite Device.

/opt/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in get_all_positioners()
     89     positioners = []
     90     for device in devices:
---> 91         positioners.extend(_recursive_positioner_search(device))
     92     return positioners
     93

/opt/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in _recursive_positioner_search(device)
     97     # TODO Refactor this as a method on Device.
     98     res = []
---> 99     if hasattr(device, 'position'):  # duck-typed as a Positioner
    100         res.append(device)
    101     if isinstance(device, Device):  # only Devices have `_signals`

/opt/conda_envs/collection/lib/python3.4/site-packages/ophyd/utils/epics_pvs.py in wrapper(self, *args, **kwargs)
    342             return fcn(self, *args, **kwargs)
    343         else:
--> 344             raise DisconnectedError('{} is not connected'.format(self.name))
    345     return wrapper
    346

DisconnectedError: bimy is not connected
```

Just to make sure, it works with connected motors.
```python
In [10]: wh_pos(assa)

+------------+----------+-----------+------------+
| Positioner |    Value | Low Limit | High Limit |
+------------+----------+-----------+------------+
| aSSA_y     |  0.45600 |   0.00000 |    0.00000 |
| aSSA_x     |  2.24650 |   0.00000 |    0.00000 |
| aSSA_dy    | 17.20400 |   0.00000 |    0.00000 |
| aSSA_dx    |  9.52990 |   0.00000 |    0.00000 |
+------------+----------+-----------+------------+
```

hhslepicka, 2016-02-03:
I can also see this error with get_all_positioners().

dchabot, 2016-02-03:
Thanks @hhslepicka. The problem is [here](https://github.com/NSLS-II/ophyd/blob/master/ophyd/commands.py#L99), and the fact that the `position` property has a `raise_if_disconnected` decorator.

I know this case was covered previously so, sorry about the regression.

I will look at this tonight.

hhslepicka, 2016-02-03:
No problem! Thanks @dchabot :+1:

klauer, 2016-02-03:
Closed by #257

closed 2016-02-03 by klauer