bluesky/ophyd #260: Problem accessing scan images

vivekthampy, 2016-02-09:
Trying to retrieve images as below: 

ims = get_images(db['e54e9e'], 'fccd_image')

gives the error:

OSError: Unable to open file (Unable to open file: name = '/gpfs/xf23id/xf23id1/fccd_data/2016/02/07/d8710b44-8f0d-498a-b0d8_-000001.h5', errno = 2, error message = 'no such file or directory', flags = 0, o_flags = 0)

The h5 file in the GPFS drive is stored as "d8710b44-8f0d-498a-b0d8_-000000.h5", and not "d8710b44-8f0d-498a-b0d8_-000001.h5".

tacaswell, 2016-02-10:
@stuwilkins Looks like that race condition is back in the stage.

klauer, 2016-03-03:
Is this resolved?

klauer, 2016-05-16:
Closing as this is not strictly an ophyd issue. Please re-open in NSLS-II/bug-reports if it still persists.

closed 2016-05-16 by klauer