bluesky/ophyd #273: Positioner/EpicsMotor Additions

hhslepicka, 2016-02-18:
@danielballan, @dchabot, @klauer 
Per discussed on the meeting on Feb, 17th regarding the EpicsMotor we decided that Positioner/EpicsMotor should implement the following:

- homing
- calibration (exists in commands.py?)
- limit switch stat (+ exception raising + status object stuff)
- velocity
- acceleration
- anything else?

danielballan, 2016-02-18:
Issue #196 reminded me that I am interested in storing `offset` (forgive me if I'm forgetting the proper EPICS name). But I don't think we need to store the raw value per event, just the offset in configuration. Would that be OK with everyone?

klauer, 2016-02-18:
@danielballan :+1: for adding `offset` to the list for configuration attrs.

hhslepicka, 2016-02-19:
@danielballan :+1: for to ```offset``` too.

hhslepicka, 2016-03-02:
Almost there! :8ball:

PR #286 (2016-03-08): ENH: Positioner/EpicsMotor Additions Partially Fix Issue: #273
- [x] Homing
- [x] Calibration - set_current_position
- [x] Limit Switch Status
- [x] Velocity
- [x] Acceleration
- [x] Offset and Offset Freeze
- [x] Check the limit switches on the _move_changed prior to invoke _done_moving.
- [x] Tests

Closes #273 
Closes #100
closed 2016-03-08 by danielballan