bluesky/ophyd #279: settle_time for custom motors class

ambarb, 2016-02-25:
Tried to increase settle_time for a motor due to noise in scan from 0.5 to 1.0 but scan time did not roughly double (31.6 s for 21 pts to 32.0 s for 21 pts).  Unable to confirm if change was effective because ````Nanopositioner.bz.describe```` is not an option as it is for ````theta.describe````.

````
In [104]: dscan(nanop.bz,-.1,.1,20)
+------------+--------------+-------------------------+--------------------------+--------------------------+
|   seq_num  |        time  |  dif_beam_stats2_total  |  nanop_bz_user_readback  |  nanop_bz_user_setpoint  |
+------------+--------------+-------------------------+--------------------------+--------------------------+
Transient Scan ID: 59052
Persistent Unique Scan ID: 'c7d647b8-52d1-4db1-8683-6695cc9fb47b'
|         1  |  23:49:10.1  |               1.77e+07  |                   -2.28  |                   -2.28  |
|         2  |  23:49:11.7  |               1.77e+07  |                   -2.27  |                   -2.27  |
|         3  |  23:49:13.3  |               1.77e+07  |                   -2.26  |                   -2.26  |
|         4  |  23:49:14.9  |               1.77e+07  |                   -2.25  |                   -2.25  |
|         5  |  23:49:16.5  |               1.77e+07  |                   -2.24  |                   -2.24  |
|         6  |  23:49:18.0  |               1.78e+07  |                   -2.23  |                   -2.23  |
|         7  |  23:49:19.6  |               1.78e+07  |                   -2.22  |                   -2.22  |
|         8  |  23:49:21.3  |               1.78e+07  |                   -2.21  |                   -2.21  |
|         9  |  23:49:22.9  |               1.78e+07  |                   -2.20  |                   -2.20  |
|        10  |  23:49:24.4  |               1.78e+07  |                   -2.19  |                   -2.19  |
|        11  |  23:49:26.0  |               1.78e+07  |                   -2.18  |                   -2.18  |
|        12  |  23:49:27.6  |               1.78e+07  |                   -2.17  |                   -2.17  |
|        13  |  23:49:29.3  |               1.78e+07  |                   -2.16  |                   -2.16  |
|        14  |  23:49:30.9  |               1.78e+07  |                   -2.15  |                   -2.15  |
|        15  |  23:49:32.5  |               1.78e+07  |                   -2.14  |                   -2.14  |
|        16  |  23:49:34.0  |               1.79e+07  |                   -2.13  |                   -2.13  |
|        17  |  23:49:35.6  |               1.78e+07  |                   -2.12  |                   -2.12  |
|        18  |  23:49:37.3  |               1.78e+07  |                   -2.11  |                   -2.11  |
|        19  |  23:49:38.9  |               1.78e+07  |                   -2.10  |                   -2.10  |
|        20  |  23:49:40.5  |               1.78e+07  |                   -2.09  |                   -2.09  |
|        21  |  23:49:42.1  |               1.78e+07  |                   -2.08  |                   -2.08  |
+------------+--------------+-------------------------+--------------------------+--------------------------+
Out[104]: ['c7d647b8-52d1-4db1-8683-6695cc9fb47b']

In [107]: nanop.desc
nanop.describe                nanop.describe_configuration  

In [107]: nanop.describe
Out[107]: <bound method Nanopositioner.describe of Nanopositioner(prefix='XF:23ID1-ES{Dif:Lens', name='nanop', read_attrs=['tx', 'ty', 'tz', 'bx', 'by', 'bz'], configuration_attrs=[], monitor_attrs=[])>

In [108]: nanop = Nanopositioner('XF:23ID1-ES{Dif:Lens', name='nanop',settle_time=0.5)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-108-8c57e73493a7> in <module>()
----> 1 nanop = Nanopositioner('XF:23ID1-ES{Dif:Lens', name='nanop',settle_time=0.5)

/home/xf23id1/conda_envs/collection/lib/python3.4/site-packages/ophyd/device.py in __init__(self, prefix, read_attrs, configuration_attrs, monitor_attrs, name, parent, **kwargs)
    462             name = prefix
    463 
--> 464         super().__init__(name=name, parent=parent, **kwargs)
    465 
    466         if read_attrs is None:

/home/xf23id1/conda_envs/collection/lib/python3.4/site-packages/ophyd/device.py in __init__(self, *args, **kwargs)
    337         self._staged = Staged.no
    338         self._original_vals = OrderedDict()
--> 339         super().__init__(*args, **kwargs)
    340 
    341     def trigger(self):

TypeError: __init__() got an unexpected keyword argument 'settle_time'

In [109]: class Nanopositioner(Device):
   .....:         tx = Cpt(EpicsMotor, '-Ax:TopX}Mtr', settle_time=0.5)
   .....:         ty = Cpt(EpicsMotor, '-Ax:TopY}Mtr', settle_time=0.5)
   .....:         tz = Cpt(EpicsMotor, '-Ax:TopZ}Mtr', settle_time=0.5)
   .....:         bx = Cpt(EpicsMotor, '-Ax:BtmX}Mtr', settle_time=0.5)
   .....:         by = Cpt(EpicsMotor, '-Ax:BtmY}Mtr', settle_time=0.5)
   .....:         bz = Cpt(EpicsMotor, '-Ax:BtmZ}Mtr', settle_time=1)  
   .....:     

In [110]: 

In [110]: 

In [110]: dscan(nanop.bz,-.1,.1,20)
Transient Scan ID: 59053
Persistent Unique Scan ID: 'f60cd046-c01d-41ca-95bd-ba6ee7586b8c'
+------------+--------------+-------------------------+--------------------------+--------------------------+
|   seq_num  |        time  |  dif_beam_stats2_total  |  nanop_bz_user_readback  |  nanop_bz_user_setpoint  |
+------------+--------------+-------------------------+--------------------------+--------------------------+
|         1  |  23:54:48.3  |               1.77e+07  |                   -2.28  |                   -2.28  |
|         2  |  23:54:49.9  |               1.77e+07  |                   -2.27  |                   -2.27  |
|         3  |  23:54:51.5  |               1.77e+07  |                   -2.26  |                   -2.26  |
|         4  |  23:54:53.0  |               1.77e+07  |                   -2.25  |                   -2.25  |
|         5  |  23:54:54.6  |               1.77e+07  |                   -2.24  |                   -2.24  |
|         6  |  23:54:56.1  |               1.77e+07  |                   -2.23  |                   -2.23  |
|         7  |  23:54:57.7  |               1.77e+07  |                   -2.22  |                   -2.22  |
|         8  |  23:54:59.3  |               1.77e+07  |                   -2.21  |                   -2.21  |
|         9  |  23:55:00.9  |               1.78e+07  |                   -2.20  |                   -2.20  |
|        10  |  23:55:02.5  |               1.78e+07  |                   -2.19  |                   -2.19  |
|        11  |  23:55:04.2  |               1.78e+07  |                   -2.18  |                   -2.18  |
|        12  |  23:55:05.7  |               1.78e+07  |                   -2.17  |                   -2.17  |
|        13  |  23:55:07.3  |               1.78e+07  |                   -2.16  |                   -2.16  |
|        14  |  23:55:08.9  |               1.78e+07  |                   -2.15  |                   -2.15  |
|        15  |  23:55:10.5  |               1.78e+07  |                   -2.14  |                   -2.14  |
|        16  |  23:55:12.0  |               1.77e+07  |                   -2.13  |                   -2.13  |
|        17  |  23:55:13.6  |               1.78e+07  |                   -2.12  |                   -2.12  |
|        18  |  23:55:15.1  |               1.78e+07  |                   -2.11  |                   -2.11  |
|        19  |  23:55:16.7  |               1.78e+07  |                   -2.10  |                   -2.10  |
|        20  |  23:55:18.4  |               1.78e+07  |                   -2.09  |                   -2.09  |
|        21  |  23:55:19.9  |               1.78e+07  |                   -2.08  |                   -2.08  |
+------------+--------------+-------------------------+--------------------------+--------------------------+
Out[110]: ['f60cd046-c01d-41ca-95bd-ba6ee7586b8c']

````

klauer, 2016-02-25:
`settle_time` was implemented for one of the very first versions of `ophyd.EpicsMotor`, but its functionality was removed at some point soon after that.

The newest version has removed the unimplemented parameter altogether which is why you are seeing the message.

klauer, 2016-03-03:
Closed by #283; `settle_time` is implemented on all positioners. @tacaswell will help you upgrade to this tagged version once it is automatically built.

```python
In [1]: import ophyd
In [2]: ophyd.commands.setup_ophyd()
In [3]: m = ophyd.EpicsMotor('XF:31IDA-OP{Tbl-Ax:X1}Mtr', name='m', settle_time=1.0)
In [4]: m.wait_for_connection()
In [5]: m.position
Out[5]: 0.0

In [6]: m.settle_time
Out[6]: 1.0

In [7]: m.move(0.0, wait=True)
Out[7]: MoveStatus(done=True, elapsed=1.0, success=True, settle_time=1.0)
```

closed 2016-03-03 by klauer