bluesky/ophyd #288: `mov` does not always update with final motor position

tacaswell, 2016-03-05:
It looks like there is just some delay between when the motors report that they are done (the status objects are flipped) and all of the new positions have propagated through epics/ophyd.

@cmazzoli reports the motor really is at the correct position so this is cosmetic.

Moved from https://github.com/NSLS-II/wishlist/issues/107

klauer, 2016-03-06:
Probably would be easiest just to temporarily increase `settle_time` for all `mov`ed positioners and reset it after.

klauer, 2016-03-16:
I can't come up with an intelligent and generic way to do this, beside for adding arbitrary delays or making some assumptions about the positioner deadband. Without deadband info, what is the "final position" after all?

tacaswell, 2017-08-02:
`mov` will be removed via #410, closing.

closed 2017-08-02 by tacaswell