bluesky/ophyd #298: Auto build and upload ophyd docs

ericdill, 2016-03-11:


klauer, 2016-05-16:
@ericdill what do we need to get this working?

ericdill, 2016-05-16:
time... :clock12: :clock1: :clock2: :clock3: :clock4: :clock5: :clock6: :clock7: :clock8: :clock9: :clock10: :clock11:

klauer, 2016-05-16:
👍 for the clocks. Waiting for @tacaswell to find the missing hour and correct you.

ericdill, 2016-05-16:
ah that was a trap, i didn't miss an hour :eyes:

ericdill, 2016-05-16:
Ok the real issue is that I have some CI scripts that I'm working on.  I have a PR into nsls-ii/docs that will autobuild the docs on new commits and push to github.com/nsls-ii/nsls-ii.github.io, but I want to also get it working on a different repo too.  All of the docs are pushed to the nsls-ii.github.io repo and so I want to make sure the scripts can handle the base repo (nsls-ii/docs) and all of the other repos (nsls-ii/ophyd, for example).

ericdill, 2016-05-16:
The idea being that you only need to add a few lines to the .travis.yml file to auto build and upload docs with https://github.com/nsls-ii/nsls2-ci.

so something like this:
```yaml
before_script:
  - git clone https://github.com/nsls-ii/nsls2-ci ~/scripts
  - chmod +x ~/scripts/*

install:
  - . ~/scriptsinstall-miniconda.sh

after_success:
  - pushd ~/scripts && HTML_DIR="build/html" DOCS_DIR="$TRAVIS_BUILD_DIR/docs" . do-the-docs.sh && popd
```

closed 2016-08-01 by danielballan