bluesky/ophyd #327: signal that adds software offset

tacaswell, 2016-05-23:
I suspect that this may already be possible, but noting while talking to IXS

their FE slits are owned by the accelerator, report back position is not-useful to BL coordinate system.  They would like a signal that will read the position back in a coordinate system of their choice.

klauer, 2016-05-23:
I would suggest a `DerivedSignal` with `get()` overridden to add the offset as desired. 
I don't believe a `DerivedSignalWithCalculationFunction`-style class should be necessary.

closed 2016-05-23 by tacaswell

cowanml, 2016-05-24:
On Mon, 23 May 2016, K Lauer wrote:
> I would suggest a DerivedSignal with get() overridden to add the offset as desired.
> I don't believe a DerivedSignalWithCalculationFunction-style class should be necessary.

Wouldn't this be better off as a calc record in a softioc?
No fancy upper level stuff required.
Available to all epics clients.

klauer, 2016-05-24:
@cowanml absolutely. Solve it at the right level.

dchabot, 2016-05-24:
This issue is one of policy, and not a very good one.

tl;dr
OPs does not want beamlines messing with FE slit offsets.

The offset required by ixs (and others) is available in the usual way: a motor record OFF field.
It may not be relevant, but the frontend slit top/bottom/inboard/outboard axes are virtual motors, implemented at the motor controller level. The gaps and centers are implemented on top of that with the synApps 2slit module (or whatever it's called).

The policy problem is that some in the accelerator group maintain that the offset of the frontend slits are critical configuration parameters, hence modification with official blessing only.

BTW -- hola from Michigan. Hope you're all doing well.

cowanml, 2016-05-24:
so another virtual motor or calc record or something on a beamline
controlled softioc could make everyone happy?

Thanks for chiming in, good to hear from you!
Hope you both are settling in well.
-matt

danielballan, 2016-05-27:
Hey @dchabot -- nice to see you on here again.