bluesky/ophyd #335: issue when moving energy axis by mov(energy, xxx)

imrealkaren, 2016-06-07:
`p.position`
`p.egu`

```python
BlueSky [2]: mov(energy, 8)

   energy             

   ---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-2-7c2b4c6f9d72> in <module>()
----> 1 mov(energy, 8)

/home/xf05id1/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in wrapper(*args, **kwargs)
    169                                     'an instance of {}'.format(n, t))
    170 
--> 171             f(*args, **kwargs)
    172         return wrapper
    173     return wrap

/home/xf05id1/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in mov(positioner, position)
    229             print('   ', end='')
    230             for p, prec in zip(positioner, pos_prec):
--> 231                 print_value(p.position, egu=p.egu, prec=prec)
    232             print('\n')
    233             print('\033[2A', end='')

/home/xf05id1/conda_envs/collection/lib/python3.4/site-packages/ophyd/commands.py in print_value(val, prec, egu, **kwargs)
    691 def print_value(val, prec=FMT_PREC, egu='', **kwargs):
    692     if val is not None:
--> 693         print_string('{: .{fmt}f} {}'.format(val, egu, fmt=prec), **kwargs)
    694     else:
    695         print_string('', **kwargs)

TypeError: non-empty format string passed to object.__format__

klauer, 2016-06-07:
Can you point us to the configuration of that positioner, and also tell us what p.position / p.egu is showing?

tacaswell, 2016-06-08:
https://github.com/NSLS-II-SRX/ipython_ophyd/blob/master/profile_xf05id1/startup/10-machine.py#L175

It is a `PseudoPositioner` , probably returning `(E, )`

tacaswell, 2017-08-02:
the `mov` function will be removed as part of #410 , closing.

closed 2017-08-02 by tacaswell