bluesky/ophyd #355: Document common completion reporting mechanisms from EPICS

klauer, 2016-08-04:
Request from @danielballan 

* Put completion (aka put callbacks)
* Record processing (with or without busy record)
* Done flag in a record (or field)

danielballan, 2017-09-07:
From an email I wrote for a scientist, some content that might be useful whenever we get around to this:

Unfortunately, documenting the various ways that EPICS handles "completion" is still an open issue. My (possibly flawed) understanding is:

1. Some multi-PV devices like EPICS motors have a separate "readback" and/or "done" PVs that are involved in checking for completion, and ophyd Devices that span these multi-PV devices handle coordination of that information.
2. Simple EPICS signals do not have separate readback or done PVs. Instead, they use "put completion". The EPICS client sends a message "write this value" and the EPICS server on the IOC sends back the message "OK," meaning, "OK I have written that value". This all happens through just one PV.
3. Other, less smart EPICS signals do not implement put completion. When the EPICS client sends a message, "write this value", the server immediately answers "OK", meaning, "OK I _will_ write that value" even though it hasn't actually done yet. For these, if we want to know when the writing is actually complete, our only option is to poll the server with messages to read the value and see whether it has changed yet. Again, this happens through just one PV.

By default, ophyd.EpicsSignal assumes it is talking to an EPICS server like the one in (2). It waits to get back that response ("OK I have written that value") and tells bluesky that it is ready. If it is hanging instead, then I think that the server is never sending a response or that the response is for whatever reason getting lost in network traffic (which is not too unusual, sadly).

tacaswell, 2017-09-23:
I think there is also a case 4 which is a variation on 1, but without the 'done' PV so we have to poll the readback.  I _think_ this is what the AD signal do.

klauer, 2020-11-04:
This is of benefit to no one in a 4 year old "we should do this" issue

closed 2020-11-04 by klauer