bluesky/ophyd #368: reverse search for PV names

tacaswell, 2016-12-09:
To easily map from PV (readily accessible from css screens) to objects in the ipython namespace.

klauer, 2016-12-09:
How would you map with complex `Device`s and PVs?

Say you have a motor in your namespace with Device prefix `A{B}Mtr` and the user searches for `A{B}Mtr.RBV` from their CSS screen. Do you give back the mtr_readback field or the base object? Or neither?

shroffk, 2016-12-09:
Here is an example of device to pv mapping already done for the accelerator
The mapping is done with the use of properties and tags, here we see an example of the fast corrector, which has a setpoint and readback from both horizontal and vertical correction.

[image]

They have a python project which understands how to process their properties and tags and provide a simplified API to the client ( I have added some psuedo code to show the underlying workings)

[image]

But end users can directly ask channelfinder
search:handle=setpoint&device=FM1...

This will return the 2 setpoint pv's for the device FM1...
 
search:handle=setpoint&elemField=x&device=FM1...

This will return the setpoing for the horizontal corrector of he device FM1...

tacaswell, 2016-12-09:
I do not mean it searchable from css screens, I mean that it is easy to get a PV out of the css screens.  Once we have that, this is just a (recursive) search over the ophyd objects in the name space (we already have this for `wh_pos`).

hhslepicka, 2016-12-10:
You mean for a given PV return the Ophyd objects that use it, right?

tacaswell, 2016-12-11:
@hhslepicka Yes.