bluesky/ophyd #369: Capture Dial value in wh_pos function

xiaojinghuang, 2016-12-14:
The current wh_pos function lists the motor names, the user dial values and the low and high limits. Can we add dial values in the list? We noticed these numbers are critical to recover positions. Thank you.

licode, 2016-12-15:
Sure. We will work on it.

tacaswell, 2017-01-04:
According to the motor record docs, the user and dial are related via

```
userVAL = DialVAL * DIR + OFFset 
```

with `DIR` being (-/+1) http://www.aps.anl.gov/bcda/synApps/motor/R6-9/motorRecord.html.

We currently have a signal on the motor record for the offset, but not the direction.  An issue with adding a dial column to `wh_pos` is that only motor records (to my understanding) have it but the notion of 'positioner' is much wider (and this is why that column is not already there).   What would you expect to see in that case?

I will have  PR shortly to add the missing signal.

klauer, 2017-01-04:
I'd guess in the case of the dial not existing, you either show the user position (or nothing at all).
It is shoe-horning motor-record specific stuff into what should be `Positioner`-only code, in my opinion. 

The extra verbose output may put off some other users who are used to the current table, as well. Perhaps you could make a separate, more verbose `wh_pos` function or a flag to the current one that defaults to being disabled.

danielballan, 2017-02-27:
To summarize: #373 started collecting/exposing these values to Python. We recommend users roll their own `wh_pos`-like function if they want to display more than the basics, as explained in the comment above.

closed 2017-02-27 by danielballan