bluesky/ophyd #374: Should we make it easier to override a signal's precision?

danielballan, 2017-01-04:
I just wrote this for SMI:

```python
class EpicsSignalOverridePrec(EpicsSignal):
    def __init__(self, *args, precision, **kwargs):
        self._precision = precision

    @property
    def precision(self):
        "override EpicsSignal.precision which draws on EpicsSignal._read_pv.precision"
        return self._precision
```

The rationale is that SMI does have have control over the relevant IOC (the undulator) which wrongly reports a precision of 0. I still think the correct solution is to fix the IOC, but even if that's doable it sound like it would take awhile to make it happen.

Should we make this kind of thing easier by incorporating this logic in `EpicsSignal` itself? I can see the case for and against -- no strong opinion here.

klauer, 2017-01-04:
It's pretty rare that you don't have write access to the `PREC` field. The inability to change it on your end comes from the channel access gateway settings that separate the accelerator and the beamline. It's arguable that restricting that field is the right choice.

I think solving it generically in `EpicsSignal` may only lead to confusion due to a lack of synchronization with the underlying record's precision field.

danielballan, 2017-01-04:
@klauer Thanks for the input. That was my instinct as well. Solving it via subclass is annoying, but that's probably a feature -- a reminder that this should be fixed in the correct place.