bluesky/ophyd #379: Add way to specify 'root' for filestore Resource.

danielballan, 2017-01-26:
The filestore <-> AD plugin integration objects currently take in the argument

```
read_path_template='/DATA/eiger/%Y/%m/%d/'`  # for example
```

which is formatted with the current date and saved in the filestore Resource record. Currently, this gives filestore no way of knowing how much of the path is part of the 'root' (a local detail) and how much is part of the relative filepath containing semantic information to be persisted through a copying operation.

Options I can think of:

1. `read_path_template=('/DATA/eiger/', '%Y/%m/%d/')`
2. `read_path_template='/DATA/eiger/%Y/%m/%d/', root='/DATA/eiger'`

Keep in mind that we also accept a `write_path_template`, an optional kwargs for when the path that the detector writing the files sees is different than the path that a process reading its files will see. This path doesn't have anything to do with filestore and thus doesn't need a notion of 'root'. It would be nice to keep those consistent: that is, it would be slightly weird if `read_*` took a tuple while `write_*` continued to take a string. Therefore I slightly prefer option (2). Obviously we will do init-time validation checking that `root` is indeed the first part of `read_path_template`.

Thoughts?

tacaswell, 2017-01-26:
`root` could also be an integer for the number of level to consider as the root.

danielballan, 2017-01-26:
Also a valid option. I think yours is more efficient and mine is more self-explaining. Either would require validation.

danielballan, 2017-01-26:
Tie-break, anyone?

hhslepicka, 2017-01-26:
I vote for number #2 by being a more self-explaining idea. :+1:

tacaswell, 2017-04-25:
go with internally concatenating root + read_path_template

danielballan, 2017-08-02:
This was done.

closed 2017-08-02 by danielballan