bluesky/ophyd #382: Data storage server down

xiaojinghuang, 2017-02-01:
This is a urgent issue. The GPFS server is down. Bluesky scan engine couldn't access to storage directory, we cannot run any scan now.

tacaswell, 2017-02-01:
Can you access the GPFS server via any other method on the workstation?  Can you IOCs see the GPFS?

My guess is not as nothing at the python layer knows about GPFS, it is simply a path in the file system.  I also do not think that the RE directly writes to the GPFS, the detector IOCs should be doing that them selves.

You should contact someone from Rob's group about this.

cowanml, 2017-02-01:
> thread.[AFOf42aiYfuvF1gwMMQSf8qc_mAhm_qrks5rYOBdgaJpZM4L0TEi.gif]

Petkus is aware of the problem and is working on it.
-matt

xiaojinghuang, 2017-02-01:
The server is back online now. Thanks a lot.

closed 2017-02-01 by tacaswell