bluesky/ophyd #398: Add support for newer AreaDetector records/plugins

klauer, 2017-05-03:
With the release of R3-0, there are some new [plugins](http://cars9.uchicago.edu/software/epics/pluginDoc.html) that should be added.

When I did the initial creation of the areaDetector classes, I compared the databases of early R2 releases (R2-0 and R2-1 maybe?) and R1-9 and didn't find any significant differences in the available records. Now, for example, from version areaDetector R2-2 on, FilePlugins have a `CreateDirectory` record. 

It's a bit clunky, but perhaps a set of mixin classes for these could suffice since NSLS-II likely has many R1-9 IOCs running.

tacaswell, 2017-05-05:
This seems like a job for recordwhat!

klauer, 2017-05-05:
Or perhaps ophyd [itself](https://github.com/NSLS-II/ophyd/blob/master/ophyd/areadetector/util.py#L124)...

danielballan, 2017-08-02:
Any chance we can bait @prjemian or @hhslepicka into taking a look at this issue? NSLS-II is still working with older versions of AD so it would be hard for us to test.

prjemian, 2017-08-02:
That's reasonable.  Certainly a couple things will come up soon to be contributed.  In a branch.

prjemian, 2017-08-02:
About to start with a Point Grey camera using AD 2.4

hhslepicka, 2017-08-02:
Here at SLAC we are using AD 2.6.. I can for sure look into that too...

closed 2020-07-10 by klauer