bluesky/ophyd #399: Add another pseudoexample to docs

danielballan, 2017-05-08:
I found this random script while cleaning up my laptop. It might be worth adding to the pseudopositioner docs. I'm not sure how different this is from examples that we already have.

```python
from ophyd import (EpicsMotor, PseudoPositioner, PseudoSingle)
from ophyd import Component as Cpt

class SamplePositioner(PseudoPositioner):
    '''
    Maintains an offset between a master/slave set of positioners
    such that the slave movement is the negative of the master's relative
    motion (i.e. maintains constant, negative, relative offset).
    Assumes that the user has adjusted the axes to their initial positions.

    Example
    -------------
    psamp_x = SamplePositioner(prefix='', name='psamp_x', concurrent=True)
    '''
    physical_sample_holder = Cpt(EpicsMotor, '...')
    beamstop = Cpt(EpicsMotor, '...')
    sample_holder = Cpt(PseudoSingle, '...')

    def forward(self, pos):
        "pos is a self.PseudoPosition"
        delta = pos - pos.sample_holder.pos
        return self.RealPosition(physical_sample_holder=pos.sample_holder,
                                 beamstop=self.beamstop.position - delta)

    def reverse(self, pos):
        "pos is self.RealPosition"
        return self.PsuedoPosition(sample_holder=pos.physical_sample_holder)

psamp_x = SamplePositioner(prefix='', name='psamp_x', concurrent=True)
```

ordirules, 2017-06-14:
Here is my poor attempt at getting a `PseudoPositioner` working (further below), and it's failing.

What I'm trying to accomplish is:
 1. I have one sample stage located on top of another. I am defining a pseudo motor for the bottom stage (`sam`) so that when I move it, the upper stage (`amp`) must be moved back to compensate for it. This pseudo motor should only take one position as input (`sam_px`), the pseudo position of the sample. 
 2. I do not have access to the motors yet so I need to emulate them. I am using `SoftPositioner`

Does someone have advice as to what I may be doing wrong? Documentation is scarce, so it's hard to figure out. Thanks! :-)

```python
from ophyd import SoftPositioner

from ophyd import (PseudoPositioner, PseudoSingle, EpicsMotor)
from ophyd import (Component as Cpt)
from ophyd.pseudopos import (pseudo_position_argument,
                             real_position_argument)

class PseudoSam(PseudoPositioner):
    # The pseudo positioner axes:
    # amp sits on top of sam
    # since we only move sam, we only need the pseudo pos of sam
    sam_px = Cpt(PseudoSingle, limits=(-10, 10))

    def __init__(self, *args, sam=None, amp=None, **kwargs):
        # the sample arm
        self.sam_rx = sam
        # the reference arm
        self.amp_rx = amp

        super(PseudoPositioner, self).__init__(*args, **kwargs)

    # decorator ensures we get named tuple
    @pseudo_position_argument
    def forward(self, pseudo_pos):
        '''Run a forward (pseudo -> real) calculation
            When sample moves by a bit, need to move amp back by same distance
            pseudo pos is one-to-one with sample arm
        '''
        # calculate the motor changes for the sample and amplifier holder each
        delta_x = pseudo_pos.sam_px.position - self.sam_px.position

        print("Moving amp back by a delta")

        # one to one between rx and px for sam
        new_sam_rx = pseudo_pos.sam_px.position
        new_amp_rx = self.amp_rx.position - delta_x
        return self.RealPosition(sam_rx=new_sam_rx,
                                 amp_rx=new_amp_rx,
                                 )
    @real_position_argument
    def inverse(self, real_pos):
        '''Run an inverse (real -> pseudo) calculation
        This is the inverse calculation.'''
        new_sam_px = real_pos.sam_rx.position

        return self.PseudoPosition(samp_px=new_sam_px)

## create fake positioners
low, high = -10, 10
samx = Cpt(SoftPositioner,limits=(low, high), source='software', egu='mm')
ampx = Cpt(SoftPositioner,limits=(low, high), source='software', egu='mm')

psam = PseudoSam(prefix='', name="sample", sam=samx, amp=ampx)
```

There are a few other questions I'll have after this. One particular one is that I don't want to move the amplifier by the requested amount to move by, but rather the actual amount the positioner has moved. The amplifier stage has a little bit more accuracy in reaching its position than the sample. But that can be left after this is resolved. Thanks!

klauer, 2017-06-15:
@danielballan's example is not quite correct. For the `RealPosition` and `PseudoPosition` namedtuples , the `real_or_pseudo_position.motor_name` is not a `Positioner` but just a position. This means you should not be doing `.position` here:
```python
new_sam_rx = pseudo_pos.sam_px.position
```

And you cannot bring in `Positioner`s from outside the class definition and expect them to work. That is, they have to be `Component`s in the `PseudoPositioner` class definition.  This is all described in the [documentation](http://nsls-ii.github.io/ophyd/positioners.html#pseudopositioner).

tacaswell, 2017-06-15:
```python

from ophyd import SoftPositioner

from ophyd import (PseudoPositioner, PseudoSingle, EpicsMotor)
from ophyd import (Component as Cpt)
from ophyd.pseudopos import (pseudo_position_argument,
                             real_position_argument)

class PseudoSam(PseudoPositioner):
    # The pseudo positioner axes:
    # amp sits on top of sam
    # since we only move sam, we only need the pseudo pos of sam
    sam_px = Cpt(PseudoSingle, limits=(-10, 10))

    sam_rx = Cpt(SoftPositioner, limits=(-10, 10))
    amp_rx = Cpt(SoftPositioner, limits=(-10, 10))

    # decorator ensures we get named tuple
    @pseudo_position_argument
    def forward(self, pseudo_pos):
        '''Run a forward (pseudo -> real) calculation
            When sample moves by a bit, need to move amp back by same distance
            pseudo pos is one-to-one with sample arm
        '''
        # calculate the motor changes for the sample and amplifier holder each
        delta_x = pseudo_pos.sam_px - self.sam_px.position

        print("Moving amp back by a delta")

        # one to one between rx and px for sam
        new_sam_rx = pseudo_pos.sam_px
        new_amp_rx = self.amp_rx.position - delta_x
        return self.RealPosition(sam_rx=new_sam_rx,
                                 amp_rx=new_amp_rx,
                                 )
    @real_position_argument
    def inverse(self, real_pos):
        '''Run an inverse (real -> pseudo) calculation
        This is the inverse calculation.'''
        new_sam_px = real_pos.sam_rx

        return self.PseudoPosition(sam_px=new_sam_px)

psam = PseudoSam(prefix='', name="sample")
# this is needed becuase they init at None
psam.sam_rx.set(0)
psam.amp_rx.set(0)
```

should work.

ordirules, 2017-06-15:
that works perfect thanks!

ordirules, 2017-06-15:
should this be an example for the docs?

I would also suggest adding some text describing what is allowed and what is not.
I didn't know that the objects were metaclassed and I think that added some confusion. thanks!

tacaswell, 2017-06-16:
Also see https://nsls-ii.github.io/ophyd/device-overview.html#module-ophyd.device which is still a bit of a work-in-progress.

ordirules, 2017-06-16:
Is there a way to create a virtual EPICS connection?
I want to create two `PseudoPositioners`, a bottom and top stage and have references to some virtual motors that are linked. I do not want to interface with EPICS yet as I don't want to waste time stopping experiments to test this.

If I understand correctly (making a lot of guesses here...), the way the class is currently defined requires you to give a constructor and info on how to construct it. For EPICS variables, I guess it's ok since you supply a PV. However, for `SoftPositioners`, it creates a fresh new instance.

Another solution could be to add a `name` parameter to the SoftPositioner and internally store a dictionary of names. You could then have two `SoftPositioner`s in two separate `PseudoPositioner` classes linked. This can be implemented by either make `SoftPositioner` a function that returns a class or metaclass it. I think this could be nice as it would allow users to test their own complex set of motors without having to worry about EPICS yet.
Would this be good? I could submit a PR suggestion as well for further discussion. Please let me know if there is a much simpler method. Thanks for the support! :-)

PS: I can also PR a suggestion to docs summarizing this page if that helps save time.

danielballan, 2017-06-20:
> If I understand correctly (making a lot of guesses here...), the way the class is currently defined requires you to give a constructor and info on how to construct it.

You're describing the arguments to `Component` here, right? If so, yes: `Component` takes a class and that args to be passed to that class's `__init__` method.

I'm not following the next paragraph. A quick example or a PR, whichever is easier, would help. Thanks!

ordirules, 2017-06-20:
Thanks. I meant `SoftPositioner` having something like a pv just like how EPICS motors are handled.
For ex:

```python
mymotor = SoftPositioner(limits=(-10,10),soft_id=10)

class Pseudo1(PseudoPositioner):
    mt1 = Component(mymotor)
    # ...

class Pseudo2(PseudoPositioner):
    mt2 = Component(mymotor)
    #...
```

So if I create a pseudo motor from an instance of `Pseudo1` or `Pseudo2` and I move `mt1` or `mt2` (in either instance), it moves `mymotor`.

klauer, 2017-06-20:
You could subclass `SoftPositioner` and make it a singleton, which would allow you to use the same instance from both `PseudoPositioner`s. (`SoftPositioner` probably would be able to handle that usage...)

tacaswell, 2017-06-22:
Probably just add 

```python
# modulo it's actually args
def __call__(self):
    # validate they are consistent?
    return self
```

The names would have to be the same though.

To get around that 

```python
def __call__(self, *args):
     other = self.__class__(*args)
     # probably implemented via the ophydobj callbacks
     self.bind_to(other)
     return other
```

would also work (but you would have to implement `bind_to`.