bluesky/ophyd #412: Ship filled-out detector classes

danielballan, 2017-08-02:
Maybe Prosilica gets renamed ProsicilaBase and Proscilica comes with all the standard plugin components by default. Some things to address to make this work:

* It has to become possible to enable and disable plugins at runtime. (cc @hhslepicka)
* We have to pick a FileStore mixin. Let's remove the "bulk-insert" variants because they were a bad idea anyway. (They put filestore and metadatastore out of the sync and should be replaced by flyers.) That only leaves one filestore mixin per plugin. Everything else (`write_path_template`, `fs`, `root`, etc.) can be set at runtime.

hhslepicka, 2017-08-02:
Thank you @danielballan for addressing this issue! If we can enable/disable the plugins at runtime all the areadetector base classes can be shipped with all the plugins but all of them disabled! ;)

danielballan, 2017-08-02:
Note to self: for enable/disable hook, the work in https://github.com/NSLS-II/ophyd/pull/326 is probably reusable.