bluesky/ophyd #428: Update usage of networkx to work with their 2.0 release

danielballan, 2017-08-19:
I'm not sure how hard it is to support both versions of networkx. Maybe we can require 2.0 for the 1.0 release of ophyd.

http://networkx.readthedocs.io/en/latest/release/migration_guide_from_1.x_to_2.0.html

jrmlhermitte, 2018-05-03:
The [changes](https://networkx.github.io/documentation/stable/release/migration_guide_from_1.x_to_2.0.html) are not that bad. I only see three lines of code that would be affected in `areadetector/base.py`: [here](https://github.com/NSLS-II/ophyd/blob/master/ophyd/areadetector/base.py#L229), [here](https://github.com/NSLS-II/ophyd/blob/master/ophyd/areadetector/base.py#L252) and [here](https://github.com/NSLS-II/ophyd/blob/master/ophyd/areadetector/base.py#L267).

As far as I can tell, these look okay since the main changes are a switch from lists to views for nodes (so updating a graph after creating the view still contains the updated graph). I don't see any need to change the code.

If that's right, then I think we just need to:
- update [lightsource recipes](https://github.com/NSLS-II/lightsource2-recipes/blob/master/recipes-tag/ophyd/meta.yaml#L21)
- update [ophyd requirements](https://github.com/NSLS-II/ophyd/blob/master/requirements.txt#L1)
- update [.travis.yml](https://github.com/NSLS-II/ophyd/blob/master/.travis.yml)

Does that sound about right?

tacaswell, 2018-05-03:
If we aren't pinning does that mean we are supporting networkx2 already?

tacaswell, 2018-05-03:
or put another way, what version of networkx do we install on travis currently?

CJ-Wright, 2018-05-04:
From your most recent commit on master you are running against networkx 2.1 https://travis-ci.org/NSLS-II/ophyd/jobs/374059268#L2004

jrmlhermitte, 2018-05-08:
closed by #533

closed 2018-05-08 by jrmlhermitte