bluesky/ophyd #449: EpicsMotor alarm fails to trigger FailedStatus

danielballan, 2017-10-18:
We can see (using caproto!) that the motor alarm goes high but ophyd doesn't raise FailedStats. Need more interactive debugging at SIX to sort this out.

tacaswell, 2017-10-26:
What PV is the alarm on?

We only check the alarm state on the user readback 

https://github.com/NSLS-II/ophyd/blob/f1c0710217e3c95f5eb9cb700e1ef9fd66b065b5/ophyd/epics_motor.py#L253-L260

hhslepicka, 2017-10-26:
@danielballan Would you mind describing what is the situation that causes it?
Also, as @tacaswell which PV is the one that you mentioned?
https://www3.aps.anl.gov/bcda/synApps/motor/R6-9/motorRecord.html#Fields_alarm

danielballan, 2017-10-27:
1. Cut power to motor.
2. Ask ophyd to move motor.
3. Ophyd reports success.

https://www3.aps.anl.gov/bcda/synApps/motor/R6-9/motorRecord.html#Fields_alarm

hhslepicka, 2017-10-27:
Cut power to motor you mean turn off the motor controller? Shutdown the
IOC? What is the case?
Which alarm do you see raised?

danielballan, 2017-10-27:
The controller. The IOC stays online.

tacaswell, 2018-05-10:
We need access to a real motor controller / IOC to work on this, punting to next cycle.

ambarb, 2019-11-19:
anybody interested in picking this back up? we can schedule some time to test during the shutdown at the beamline this was found at.

danielballan, 2019-11-20:
Tentatively yes. We are struggling to use current ophyd `master` in production, period, and we need to resolve that first before we can build changes like this one on top. Can we schedule the time and use it both to take ophyd `master` through its paces and (if that goes well) work on alarms?